Ultimaker/Cura #14278: Wrong Machine Prefix in Filename since 5.21 update

Polly-Jeanius, 2023-01-12:
### Application Version

5.2.1

### Platform

Mac OS 12.6 

### Printer

Ender 3

### Reproduction steps

For 2.5 years I've been using Cura with my Ender 3. It's just an Ender 3, not a V2, Pro or S1.  I've always had the 'Add Machine Prefix to Job Name' setting ticked and every single file I've sliced and exported has had the CE3 prefix. Since upgrading to 5.21 this morning every subsequent file I've sliced today spits out a 'CE3E3V2'  prefix instead of the usual 'CE3'. 

### Actual results

A lengthy and inaccurate machine prefix is in my sliced file names, unless I stop and edit out the extra 'E3V2' during save. 

### Expected results

Just the usual 'CE3' prefix should be added (as it has for the last 2.5years) to my sliced files on save. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fieldOfView, 2023-01-12:
If all else fails, you can install the Printjob Naming plugin from the Marketplace and set the prefix to anything you want.

Polly-Jeanius, 2023-01-12:
Old prefix back in place after turning off the 'Add Machine Prefix to Job Name' Cura setting and using the Printjob Naming plugin instead. Nice work-around. Thanks again for the fast and helpful reply @fieldOfView!

MariMakes, 2023-01-16:
Hey @Polly-Jeanius,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Sorry, it took a while to get back to you, I'm happy to see the workaround form @fieldOfView is working for you. 

What you are seeing is intended and seems to be introduced in July 2022 to include the Ender-3 v2 since they seem to use the same settings. https://github.com/Ultimaker/Cura/pull/12442

Since you have a solution in place, it doesn't seem to be an issue anymore. 
Would you still like me to keep this issue open or do you agree that we can close it?

closed 2023-01-31 by github-actions[bot]