Ultimaker/Cura #14279: Bambu X1C printer config

czaras, 2023-01-12:
### Is your feature request related to a problem?

When r u going to add Bambu to printer list?

### Describe the solution you'd like

When r u going to add Bambu to printer list?

### Describe alternatives you've considered

manual enter

### Affected users and/or printers

Bambu

npoulos, 2023-01-12:
manual config details would be good for now!

czaras, 2023-01-13:
> Thanks for the inquiry.
> The printer definitions come from either a member of the community or from a printer manufacturer. They are submitted to UltiMaker for inclusion in Cura. Ultimaker has no way to provide definitions for other manufacturers equipment.
> There is information HERE <https://github.com/Ultimaker/Cura/wiki/Adding-new-machine-profiles-to-Cura> on how to add a printer. In the meantime you can use the Bambu slicer which (at a glance) seems to be capable.

Thx, yes, I'm using Bambu slicer but for a few things I still prefer Cura slicer ;)

regards
Jan

GregValiant, 2023-01-13:
I installed the Bamboo slicer.  From what I see in the gcode it is a lot different than Cura and plain Marlin.
You can try to use a Bamboo Gcode to set up a "StartUp Gcode" and "Ending" gcode in Cura.  The opening statements of a Bamboo file are about 930 lines of code.  You have your work cut out for you.

There seem to be a lot of added commands in the Bamboo gcode and movement commands are split up into separate lines.  I don't know if you can get that to work.

JBlakeMCAA, 2023-02-20:
Has anyone added Bambu Carbon X1 the list yet?

GregValiant, 2023-02-20:
I don't think you can get Cura to work with Bambu printers unless you make a firmware change.
This is a snippet from a Bambu gcode file.  It's wildly different than what Cura would generate.

; WIPE_END
G1 E-.012 F1800
; layer num/total_layer_count: 13/255
M622.1 S1 ; for prev firware, default turned on
M1002 judge_flag timelapse_record_flag
M622 J1
 ; timelapse without wipe tower
M971 S11 C10 O0

M623
G17
G3 Z2.8 I.944 J.768 P1  F28800
G1 X119.928 Y143.082 Z2.8
G1 Z2.6
G1 E.8 F1800
; FEATURE: Inner wall
; LINE_WIDTH: 0.45
G1 F7076

Kreyren, 2023-05-08:
I created what i know in https://github.com/Ultimaker/Cura/pull/15474 and will research what needs to be done for it to be mergable now, feel free to join on the work

Kreyren, 2023-05-08:
> I don't think you can get Cura to work with Bambu printers unless you make a firmware change. This is a snippet from a Bambu gcode file. It's wildly different than what Cura would generate. -- @GregValiant (https://github.com/Ultimaker/Cura/issues/14279#issuecomment-1436261026)

Can you give me the start and end gcode so that i can add that to overrides for machine_{start,end}_gcode in https://github.com/Ultimaker/Cura/pull/15474 ? *they don't want to let my fingers anywhere near the printer in makerspace atm to get them myself xD*

GregValiant, 2023-05-08:
Good luck.
[3DBenchy_PLA_48m58s.zip](https://github.com/Ultimaker/Cura/files/11425258/3DBenchy_PLA_48m58s.zip)

GregValiant, 2023-05-09:
I post-processed this one.  I changed all the "; CHANGE LAYER" lines to ";LAYER:#" and then threw out all the other comment lines.  There is still a lot of rambling going on (";LAYER:0" is line 552 in the file) but I think it's easier to read.  
[3DBenchySpFish_Bamboo_PostProc.zip](https://github.com/Ultimaker/Cura/files/11426487/3DBenchySpFish_Bamboo_PostProc.zip)

odie5533, 2023-07-15:
I would like to offer a $150 bounty to get a Bambu Lab X1C or P1P profile merged into Cura so it's available for general use to be added as a supported Printer. For me, AMS multi-color support and wifi printing is not needed, but fans and such must work. Please note your intention to work toward the bounty.

GregValiant, 2023-07-15:
It will take longer than an hour to do and the author will need a printer.