Ultimaker/Cura #14280: Start and End G-Code reset again

XeroZar, 2023-01-12:
### Application Version

5.2.1

### Platform

Windows 10 Pro

### Printer

Ender 3 Pro

### Reproduction steps

Opened an older stl file, everything seem fine, no issues.

### Actual results

Generate gcode and print, but noticed that the startup gcode had changed again. Checked  Cura settings and it had changed back to the default.

### Expected results

Opening any older gcode file should not affect any settings in cura in this case startup and end gcodes.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[ChainToBed.zip](https://github.com/Ultimaker/Cura/files/10403863/ChainToBed.zip)
This was the file this time. But this issue has had happen before.

GregValiant, 2023-01-12:
Thanks for the report.
I don't think that can happen with a gcode file.  The gcode reader is a separate plugin from the main Cura app.  In addition the current Ender3 Pro definition file has Cura keywords in it.  The gcode file would have numbers.  There would need to be a translator of some sort.  That type of functionality doesn't exist in Cura.

If you open a 3mf Project File then yes, your StartUp and Ending Gcode will be over-written if you allow the 3mf to "Update" your printer.  That's one of the purposes of a Project File.
I open a **_LOT_** of gcodes and 3mf files.  For 3mf files I have to be careful to select "Create New" rather than allowing them to update my Ender 3 Pro and over-write my custom start and end gcodes.
I keep a Project File of a calibration cube and it has all my defaults.  That way if an over-write does occur I can open that default file and set everything back as I want it.

I'll leave the bug label on this for now, but I really don't see opening a gcode causing the problem.  It's just a reader, there is no other functionality.

XeroZar, 2023-01-14:
> Thanks for the report. I don't think that can happen with a gcode file. The gcode reader is a separate plugin from the main Cura app. In addition the current Ender3 Pro definition file has Cura [...]

Thank you for the reply... but both times it was the STL file.

Basically i think the issue is before I installed version 5 beta, I had all versions of cura installed. After version 5.1 I had issues to which I uninstalled all the previous versions. I had also changed my start and end gcodes at this point. So is there anywhere I can specially check and remove old cura settings cause I've checked everywhere.

GregValiant, 2023-01-14:
In Windows you can go to
\Users\YourAccount\AppData\Roaming\cura\
That is where your local data is kept for each version.  I found my printer under "definition changes" and the startup gcode was in a .config file.
The first line of my StartUp gcode is ";  Greg's StartUp".  That way I can tell if it has been overwritten without having to look at every line to figure out what might be different.

XeroZar, 2023-01-14:
> In Windows you can go to
>
> \Users\YourAccount\AppData\Roaming\cura\
>
> That is where your local data is kept for each version.  I found my printer under "definition changes" and the startup gcode was [...]

I have already done that... I even uninstalled cuts and reinstalled