Ultimaker/Cura #14281: Summary: Everything wrong with Bridgeing settings

Infraviored, 2023-01-12:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Hypercube Evolution

### Reproduction steps

1. Sliced a specific model using Bridgeing settings
2. analyze different errors

### Actual results

**Bug 1)**
When selecting Wall Ordering from inside to outside and the print walls are thinner than the wall width (wall count * nozzlediameter) selected, bridgeing will start in mid-air, causing the print to fail completely (overexaggerated example with 8 walls):
(1 picture, 1 3MF file)

**Bug 2)**
With wall ordering from outside to inside, the wall lines of the bridge are printed fine (as they have supportive "real" wall lines from the outside).
However, the overhang (bridge without wall) fails on one side (2 pictures, 3MF)
This is printed with 4 walls.
To make sure no other setting is causing this, i completely disabled coasting and linear advance, and set an infill overlap percentage of 50.0% 
Still, the bridges do not connect on one side, but perfectly on the other.
From the layer view in cura, i can not see why the left wall would work while the right one does not.

### Expected results

First lines of the Bridge NEVER in mid-air, also not when high wall count and inside to outside ordering.

Normal retraction, not premature one causing the bridges to fail.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

**Bridgeing Bug 1**
[image]

3MF file for reproduction:
https://community.ultimaker.com/applications/core/interface/file/attachment.php?id=96372&key=74b49cf0d7685e0e7417a0648a839cd1

**Bridgeing Bug 2**
[image]
[image]
3MF file for reproduction:
https://community.ultimaker.com/applications/core/interface/file/attachment.php?id=96373&key=080aac830523f6d84bfc7007c4b43b96

smartavionics, 2023-01-13:
"bug 1" isn't really a bridging bug. Bridging is all about filling the gaps between supported end points. In your example, the inner green line isn't supported anywhere and so bridging isn't relevant. Of course, it's still a bug in the sense that the slicer could be smarter and omit the unsupported inner walls and, instead, expand the skin region so that it does bridge between the already printed walls. Unfortunately, it doesn't do that.

"bug 2" I cannot evaluate because your 3mf file is inaccessible to me and so I can't check your settings.

smartavionics, 2023-01-13:
However, I can see from your picture that you are using a skin wall and that really isn't going to help in this situation.

Infraviored, 2023-01-19:
[bridgeing_bugs.zip](https://github.com/Ultimaker/Cura/files/10456650/bridgeing_bugs.zip)
@smartavionics Here the 3MF files again so they can be downloaded directly.

@MariMakes thank you! i will look into that.

MariMakes, 2023-01-20:
Hey @Infraviored,

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it?

closed 2023-02-03 by github-actions[bot]