Ultimaker/Cura #14283: Exported GCODE files do not indicate the printer for which they were sliced

smartin015, 2023-01-12:


smartin015, 2023-01-12:
Thanks @jellespijker - and just to reiterate, I'd gladly contribute a PR if you think it's worthwhile.

smartin015, 2023-03-08:
Hey @jellespijker, @MariMakes - this is still something I'd like to have available in Cura sliced files. It's been 5 months - could I at least get a yes/no on whether I can implement this?

MariMakes, 2023-03-09:
Hey @smartin015,

Nice to meet you. 
I apologize for waiting so long to respond.
It must have gotten lost in the pile somewhere.

Do I understand correctly that you have a feature request for:
Including the Machine Definition Name in all Gcode Headers.

Because it's something we do in Griffin, which is used for our Ultimaker Printers already, like this:
[image]

Personally, I use the printjob prefix to distinguish the different printers. 
[image]
You can make them even [more elaborate with this plugin.](https://github.com/fieldOfView/Cura-CustomJobPrefix), but that might not be a solution that works for you.  

I'll bring it up to the team to see what risks they see. 
A number of the risks will also be covered with completing the pull request template correctly, https://github.com/Ultimaker/Cura/blob/main/.github/PULL_REQUEST_TEMPLATE.md
I think one of the trickiest things is learning if there are printers that will break when this is implemented. 

I'll let you know what the team thinks. 🤞 Fingers crossed.

smartin015, 2023-03-09:
Hey @MariMakes - thanks for taking a look :)

> Do I understand correctly that you have a feature request for: Including the Machine Definition Name in all Gcode Headers.

I believe so - the `TARGET_MACHINE.NAME` header field looks to be specifically what I'm after. I intend to use this string to soft match against a list of known printers, which will direct gcode files that successfully match to printers of that type on the local network. I'm not familiar with Griffin as a slicer flavor, but if this can be back-ported (?) to Marlin that'd be greatly appreciated.  

> Personally, I use the printjob prefix to distinguish the different printers. [image] You can make them even [more elaborate with this plugin.](https://github.com/fieldOfView/Cura-CustomJobPrefix), but that might not be a solution that works for you.

As print files can be easily renamed or mutated by other plugins, I don't want to rely on that for automation. I'd much prefer inspecting the file itself for metadata about what it's sliced for, especially if the metadata is included by default (no special setup needed for Cura users of my plugin).

> I'll bring it up to the team to see what risks they see. A number of the risks will also be covered with completing the pull request template correctly, [...]

AFAIK, the risk should be very low. I'm requesting adding a single commented line into a gcode file that already has commented lines, so I'd be surprised if any printer models wouldn't just skip over it when the file is executed.

> I'll let you know what the team thinks. 🤞 Fingers crossed.

Thanks!

MariMakes, 2023-03-13:
Hey @smartin015,

I've discussed your issue with the team and they like the initiative. 😄 
Most gcodes already have some comments in them, for example to mention the gcode flavor. Including the TARGET_MACHINE.NAME metadata could prove useful for a number of applications. 

So a PR is very welcome 😺 🎉 

I'm just the community manager, but I believe this is a good place to start. 
https://github.com/Ultimaker/CuraEngine/blob/cc03a5711c1a28e91cd48c1ffef19dba852b4ca1/src/gcodeExport.cpp

Again sorry for the slow responses. 
Please feel free to tag me if you have the idea that your Pull Request is drowned out again.

smartin015, 2023-03-14:
Woohoo, thanks for clearing this :) 

It'll be a couple weeks before I start working on it, but once I do I'll take a look at the file you provided and see how to plumb the data into a header comment.

smartin015, 2023-04-01:
Okay, finally had a chance to take a shot at this today - #1871. Should be a quick review, I just copied L177 to the default case so `TARGET_MACHINE.NAME` is present for all gcode flavors.

I had some trouble getting a build environment set up, as Conan installs v2.0 by default and the commands (e.g. `conan install`) are different. Pinning to the previous version (1.59.0) worked, but I also had some strange interactions with other installed libraries, so decided to create a Dockerfile to avoid all that.

Adding it to this comment as as personal reference, but I could contribute this to the [building from source](https://github.com/Ultimaker/CuraEngine/wiki/Building-CuraEngine-From-Source) guide if you'd like:

Dockerfile:

```
FROM ubuntu:22.04

RUN apt update && apt install -y git python3 ninja-build python3-pip build-essential wget \
  && mkdir /cmake && cd /cmake && wget https://github.com/Kitware/CMake/releases/download/v3.26.2/cmake-3.26.2-linux-x86_64.sh \
  && chmod a+x cmake-3.26.2-linux-x86_64.sh \
  && ./cmake-3.26.2-linux-x86_64.sh --skip-license \
  && ln -s /cmake/bin/* /usr/bin/ \
  && pip3 install conan==1.59.0 \
  && conan config install https://github.com/ultimaker/conan-config.git \
  && conan profile new default --detect --force \
  && rm -rf /var/lib/apt/lists/*
```

Commands to launch the container, mount the current (repository) directory, and build/run the test:

```
docker run -it --rm -v $(pwd):/CuraEngine curaengine /bin/bash
conan install . --build=missing --update -s build_type=Debug -o curaengine:enable_testing=True
cmake --preset debug
cmake --build --preset debug && ./build/Debug/tests/GCodeExportTest
```

smartin015, 2023-04-01:
Also the CI failure appears to be some configuration issue unrelated to the PR - all the unit tests pass, but there's a failing `Conclusion` step (see PR for screenshot)

smartin015, 2023-04-07:
Hey @MariMakes, the PR is ready for review (the CI failure appears to be unrelated to my change). 

Could you please find a reviewer? Thanks!

MariMakes, 2023-05-03:
Awesome @smartin015 😁 
I'll ask the team to take a look 💪