Ultimaker/Cura #14284: Slicing failed

Magic2226, 2023-01-13:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Creality Ender 3

### Reproduction steps

Prepaired model slicer settings and hit slice. Slice failed with tree supports. Normal zigzag supports worked.

### Actual results

Slicing failed

### Expected results

Sliced expected. 

[CE3E3V2_Ostomy cup 2023.zip](https://github.com/Ultimaker/Cura/files/10409006/CE3E3V2_Ostomy.cup.2023.zip)
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/10409077/cura.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Files attached. Normal supports work every time. Tree supports failed 3 of 5 times.

GregValiant, 2023-01-13:
Thanks for the report.
I used 5.2.1 for this.  5.3 gave the same results.  I had no trouble slicing this in either version.

This is with your tree support settings:
[image]

This is with 1 change to your setting profile.
[image]

The difference is that you have your Support Wall Line Count = 0 and so all the lines are standing alone and unsupported.  You support needs support.
That may be affecting the slice although it sliced every time for me regardless of the settings.

Although I can't duplicate the problem I'll mark this as a slicing error and someone from the Cura team will take a look.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker