Ultimaker/Cura #14285: unable to slice

69Rigs, 2023-01-13:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Pixel s1

### Reproduction steps

Slice the model

### Actual results

says the software is unable to slice

### Expected results

should have sliced it

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

issue doesnt happen when ender 3 pro is selected instead

GregValiant, 2023-01-13:
Thanks for the report.
If it slices for the Ender then maybe you have a setting that is beyond one of the "Maxes" for the Pixel.
Load the "Printer Settings" plugin from the Market Place and check things like the "Max Z Feedrate".  There are printers with Max Z speed of 5mm/sec and if you try to slice with the Zhop speed at 10mm/sec you would get the "can't slice" message.
If a setting is the issue then the setting box should be showing a red background.
A Project File would be very helpful.  Without one it's all guesswork.

69Rigs, 2023-01-13:
Thanks for the help. I'm surprised I got a response so quickly. I've
attached the stl file here. I'll test out what you suggestd when you have
time. I appreciate the help.

 skull clean.stl
<https://drive.google.com/file/d/1b5INuopYUJ7M-jFSyczoA12sX8gs-eIQ/view?usp=drive_web>
Clint Johny
69Rigs

GregValiant, 2023-01-13:
I'm retired.  It's all about when I roll out of bed and the coffee is done.
With the Pixel printer active, load the model in Cura and set Cura up to slice.  Use the "File | Save Project" command to create a project file.  It will include the model, your settings, and the Pixel S1.

From what I see of the STL file there are some extremely delicate features within the skull.  I really see no way to print this successfully.  Removing the supports would likely destroy parts of the model.
I didn't have any trouble slicing it for my Ender 3 Pro either (but you knew that).

69Rigs, 2023-01-13:
Thanks Greg - I'm really struggling here right now. I've been figuring out
how to make STLs from Dicom files right now. I'm their only option - 3D
printing - especially for these kinds of applications is almost
non-existent in India right now - they need it for medical purposes. The
patient is an old customer of mine. He had a horrible accident and his mom
called me to help his son out. I'm gonna have a sleepless night today but
let's hope for the best - at least I'm making progress. This model is a lot
cleaner now. I'm still wondering how i'll get it to print. it's a miracle
he's still alive after this. I'm just praying for a miracle with this print
too. they need 3 sections so maybe that will work out somehow.

Clint Johny

[image: image.png]