Ultimaker/Cura #14286: Slicing Failed

DarkMaster007, 2023-01-13:
### Application Version

5.2.1

### Platform

OS: Windows 11, GPU: 3060 TI

### Printer

Ender 3 Pro

### Reproduction steps

1. Downloaded .stl from https://www.thingiverse.com/thing:4885937 that has all dice in one file. 
2. File loaded into cura
3. Clicked "Slice"

### Actual results

I receive a slicing error from Cura.

### Expected results

Model should be sliced and a prompt to print or preview should appear.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

The log and project file: https://mega.nz/file/ZRZj1CRB#nL5R63ouCyZudHdy9_5xQSI9YMpYi5K2Og_Du2ZxHKQ

GregValiant, 2023-01-13:
Thanks for the report.
It's hard to see here but Cura has colored many of the numbers "Polka Dot".  That indicates errors in the model.
[image]

It's easier to see in X-ray view.
[image]

So the models need repair.  I didn't have any trouble slicing this (without repair) but I don't think I'd trust it to print correctly unless the models were repaired.  It's hard to tell if you are experiencing a slicing bug or if the models are the problem.  You can try grouping the models in Cura and rotating/moving the group on the build plate.  You may find an orientation that allows it to slice.

DarkMaster007, 2023-01-14:
Yea, that might be the problem since adding them separately and using auto arrange worked in slicing them. It also printed quite well (as well as an 0.4 allows on small details). Never had an issue slicing so I didn't think of model problems. Should I close this since it was a model issue?

GregValiant, 2023-01-14:
Yeah, let's close this one.  You got them to slice so the problem is fixed.  Whether it was a software bug or a model problem would be hard to determine.

closed 2023-01-14 by GregValiant