Ultimaker/Cura #14287: slicing error

Xiaacs, 2023-01-13:
### Application Version

5

### Platform

windows 10

### Printer

anycubic kobra neo

### Reproduction steps

i have been trying to slice and it wont let me. it keeps popping up error

### Actual results

so i try to slice and it keeps popping up slice error.

i have a anycubic kobra neo printer.

### Expected results

i need help to understand cura and to make sure it works for me lol.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

im trying to do the owl and it will not let me

GregValiant, 2023-01-13:
Thanks for the report.
@Xiaacs let us assume that I have no idea what you are talking about.
What owl?  Is it an STL owl?  A gcode owl?  Is it a live owl?
More information would be good.

Xiaacs, 2023-01-14:
Yes the stl owl that anycubic gives you to practice with

GregValiant, 2023-01-14:
Move the owl on the build plate.  You may try rotating it as well.  There is a bug that can prevent some models from slicing when they are at the Cura build plate midpoint.  You can try something from Thingiverse as well.
Let us know if that works for you.

Xiaacs, 2023-01-14:
I also did try a jack skellington coin and it did the same as well.

I did them separate but took a pic of them together to show

Xiaacs, 2023-01-14:
What am I doing wrong?

MariMakes, 2023-01-16:
Hey @Xiaacs,

Welcome to the Ultimaker Cura Github 🚀 

Is it possible that your printjob is located at 0, 0, 0, and it slices if you move it a little away from the center of the buildplate? 
Then it might be a duplicate of https://github.com/Ultimaker/Cura/issues/13643

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

MariMakes, 2023-01-17:
Hey @Xiaacs,

I really want to help you but I need the file you use for slicing.

If I pick a random Jack Skellington Keychain from Thingiverse it slices for me:
https://www.thingiverse.com/thing:1164530/files
[image]

Do you have a project file for us for the Jack Skellington model you are trying to slice? 
It contains the printer and settings we need for troubleshooting. 

Can you go to your laptop, and save a project file?
1. Go to File
2. Save project
3. Zip the .3mf project file. 
4. You'll need to open Github in your browser to attach any files to your ticket. You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

Xiaacs, 2023-01-17:
So good news, it was my laptop which was the issue, I had to do a hard reset and now it works. I am so sorry for taking your time.

https://www.thingiverse.com/thing:1164530/files
[image]<https://user-images.githubusercontent.com/40423138/212861066-375aea44-68d1-4200-aa74-6ba7b2b871de.png>

closed 2023-01-17 by GregValiant