Ultimaker/Cura #14288: Opening a Cura project file inverts Z axis for calculation of overhang preview

Jceggbert5, 2023-01-13:
### Application Version

5.3.0-alpha+xmas

### Platform

Windows 11 22H2, 22623.1095 / 1000.22638.1000.0

### Printer

Creality Ender-3 Pro

### Reproduction steps

1. Open a 3MF file
2. Pick either option (import printer profile or just the models)

### Actual results

[image]

Support/overhang preview shows on top of the model instead of the bottom. Manipulating the models does not change behavior, nor does restarting Cura. Also happens when opening on another PC (same cura version, windows 10 and 11).

### Expected results

It should show up on the bottom, to preview where support may end up.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[joints.zip](https://github.com/Ultimaker/Cura/files/10414313/joints.zip)
[feethands.zip](https://github.com/Ultimaker/Cura/files/10414319/feethands.zip)

3MF files were created using this version of Cura.

GregValiant, 2023-01-13:
Thanks for the report.
There are A LOT errors in those models.  That's why Cura is showing upper surfaces in red

The models can be repaired in MS 3D Builder.  You can tell which models I repaired.
[image]

Here are the joints repaired.
[image]

I'm going to go ahead and remove the bug label from this report and close this.  I hope you understand.
Those models have a lot of errors that are confusing the software.

closed 2023-01-13 by GregValiant

Jceggbert5, 2023-01-13:
> Thanks for the report. There are A LOT errors in those models. That's why Cura is showing upper surfaces in red
>
> The models can be repaired in MS 3D Builder. You can tell which models I repaired. [...]

If the models have errors, wouldn't they show up incorrectly during the initial import, and not wait until after save/load of the 3MF file?

Jceggbert5, 2023-01-13:
Interesting. I'm surprised it showed properly when setting up the project initially rather than after save/load.

Unrelated, can I ask what plugin(s) you're using for the more detailed Ender3 model?

GregValiant, 2023-01-13:
I design all my own stuff.  Since I'd modeled up the printer to be the base for my add-ons I just exploded it and put it together to be the platform STL in Cura.

I'm starting to wonder if that isn't just a corrupted 3mf file.  There are so many errors it's hard to believe it would even slice.
This is the view of the 3mf opened in Mesh Mixer.  The app uses magenta balls to indicate errors in a model.  They are so thick here that it looks like a sphere.  The app indicated over 25,000 errors.  Although they are centered around the left hand, I don't think it's really the problem.  Just a feeling.  If I hit the "Auto-Repair" button then the hand will disappear.
[image]