Ultimaker/Cura #14291: Loading STL after loading g-code glitches cura

Zydnar, 2023-01-14:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

unrelevant

### Reproduction steps

1. I loaded g-code instead of .stl
2. I loaded .stl file because g-code I loaded by mistake
3. Change printing settings

### Actual results

Changing settings does not bring back slice button

### Expected results

Changing settings does bring back slice button

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10417209/cura.log)

GregValiant, 2023-01-14:
Thanks for the report.  Someone from the Cura team will take a look.
I can't duplicate this on my installation of 5.2.1 or 5.3.

MariMakes, 2023-01-20:
Hey @Zydnar,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 

I can reproduce your issue. Usually you can clear the build plate or start a new project to resolve the issue, but that is not the case here You have to restart Cura to fix it😮 

It reminds me a little of https://github.com/Ultimaker/Cura/issues/14261
I think it might be related. I'll check with the team if that is indeed the case.

MariMakes, 2023-02-09:
Hey @Zydnar,

I discussed your issue with the team but forgot to update you. 

This indeed seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/14261
We have one of our developers looking at your issue as we speak, so hopefully, we can resolve it soon. 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/14261
Thanks again for you report. 👍

closed 2023-02-09 by MariMakes