Ultimaker/Cura #14292: Slice error message

Speedy-McCat, 2023-01-14:
### Application Version

5.3.0-alpha+xmas

### Platform

Windows 10

### Printer

Creality Ender 3 S1 Pro

### Reproduction steps

1. an object selected in the 3D Builder
2. divided into a 6mm big subpart
3. saved as an STL file
4. opened in Cura, there was no error message
5. Slice selected
6. Then the error message came

### Actual results

1. Then the error message came
2. the slicing was not executed

### Expected results

that the object is sliced
[CE3S1_Ender_3_S1_Pro_Camera_Mount Kamerahalter 2 V1 TEST.zip](https://github.com/Ultimaker/Cura/files/10417293/CE3S1_Ender_3_S1_Pro_Camera_Mount.Kamerahalter.2.V1.TEST.zip)
[cura_log.zip](https://github.com/Ultimaker/Cura/files/10417301/cura_log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

In the file "CE3S1_Ender_3_S1_Pro_Camera_Mount Camera Holder 2 V1 TEST.zip" is the .3mf
[image]

GregValiant, 2023-01-14:
Thanks for the report.
When I open the 3mf file the model is 0.062mm thick.  That won't slice unless you drop your layer height to .06mm and then it is only a single layer thick.
My guess here is that because no layers were generated at 0.20 layer height that Cura decided it was a bug.  This is at 0.06 layer height.  I would not be able to print this on my Ender as it is just too thin.
[image]

MariMakes, 2023-01-20:
Hey @Speedy-McCat,

Thanks for your report 👍

At first I thought your model might be in inches since it was so thin. 
But than it's way too big to print 😅 
[image]

To me, it doesn't seem like a bug in Ultimaker Cura, since the mesh cannot be printed in these dimensions anyways. 
Would you agree that we can close the issue, or would you like me to keep it open?

closed 2023-02-03 by github-actions[bot]