Ultimaker/Cura #14293: Print Seq selected but option not showing

servant74, 2023-01-14:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Ultimaker 3

### Reproduction steps

1. Searched for 'print seq'
2. checked to show Print Sequence in menu
3. Does not show in menu.

Tried the same with ArcWelder (which is installed) and it is showing correctly.

I am 'bumfuzzled' to say the least.  We also tried on 5.1.1 with the same results.

### Actual results

Print Seq never showed up as an option even after restarting Cura.

### Expected results

We were epecting the print sequence option to show to allow 'one at a time' printing

[UM3_ThinMintSprint2Color.zip](https://github.com/Ultimaker/Cura/files/10418018/UM3_ThinMintSprint2Color.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[UM3_ThinMintSprint2Color.zip](https://github.com/Ultimaker/Cura/files/10418023/UM3_ThinMintSprint2Color.zip)

GregValiant, 2023-01-14:
Thanks for the report.
"Print Sequence" is disabled for all multi-extruder printers when more than 1 extruder is enabled.
If you disable an extruder it will appear.  (There must have been a reason why it was designed this way and so this is expected behavior.)  I would think there would be no difference for "Multi-In-One-Out" hot ends, but they are the same).
I'll remove the bug label but leave this open so you can respond.

servant74, 2023-01-14:
Ok, at least I understand that is the case,  no I don't understand why but I won't spend the time to go through the code either.

Thanks for the timely response.

closed 2023-01-14 by servant74

GregValiant, 2023-01-14:
FYI - I found this in fdmprinter.def.json.
"print_sequence":
                {
                    "label": "Print Sequence",
                    "description": "Whether to print all models one layer at a time or to wait for one model to finish, before moving on to the next. One at a time mode is possible if **a) only one extruder is enabled** and b) all models are separated in such a way that the whole print head can move in between and all models are lower than the distance between the nozzle and the X/Y axes.",
                    "type": "enum",
                    "options":
                    {
                        "all_at_once": "All at Once",
                        "one_at_a_time": "One at a Time"
                    },
                    "default_value": "all_at_once",
                    **"enabled": "extruders_enabled_count == 1"**,
                    "settable_per_mesh": false,
                    "settable_per_extruder": false,
                    "settable_per_meshgroup": false
                },

servant74, 2023-01-14:
I find it interesting we are only using one extruder at a time, and never on the same level.  But still are not allowed to use this option.

GregValiant, 2023-01-14:
In multi-color prints they would be used on the same level.
This is one of those things that may have been in place for years and was never questioned.  Typically, dual extruders were large print heads that would have been hard to work around.  In the last two years the popularity of multi-in-one-out hot ends has grown and this setting may need to be revisited.  A workaround should be obvious from my previous post - true?