Ultimaker/Cura #14294: Double unretraction with "Retract at layer change" and combing "Not on outer surface". Causes oozing with timelapse.

darrabito, 2023-01-14:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Custom Printer - Ender 3 S1

### Reproduction steps

Enable settings listed.
Slice a cylinder.

### Actual results

Turn print speed super slow to watch retractions happen twice at layer change.
Observe insane amount of oozing due to no retractions before timelapse.

Problem Gcode:
`G1 F4514.7 X102.961 Y105.161 E322.94802  ; latest extrusion path
G1 X102.899 Y105.053
G0 F15000 X102.905 Y105.06
G1 F2400 E322.14802       ; RETRACT AT LAYER CHANGE
;MESH:NONMESH
G0 F300 X102.905 Y105.06 Z5.4
G0 F15000 X100.971 Y109.37
G1 F2400 E322.94802      ; UNRETRACT?
G0 F15000 X100.274 Y109.32
;TIME_ELAPSED:185.297087
TIMELAPSE_TAKE_FRAME
;LAYER:26
G1 F2400 E322.14802      ; RETRACT
;TYPE:WALL-OUTER
;MESH:Cylinder
G1 F2400 E322.94802      ; UNRETRACT`

### Expected results

Retract only once when both of those settings are enabled at layer change. Unretract after timelapse photo and new layer start.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Related:
https://community.ultimaker.com/topic/41807-cura-511-retracts-twice-between-layers/

GregValiant, 2023-01-14:
Thanks for the report.
Without a project file I can't tell how you have this set up.  From what I see in the code snippet it is a lot different than what I get using "Time Lapse".  This is the code from the last extrusion until extruding resumes.  This is with "retract at layer change" enabled and combing set to "not on outer surface" and No Retraction by Time Lapse.
G1 F4500 X104.039 Y125.535 E70.27162
          _G1 F2100 E64.27162_
;MESH:NONMESH
G0 F600 X104.039 Y125.535 Z0.4
G0 F9000 X103.6 Y125.535
G0 X103.5 Y103.5
;TIME_ELAPSED:40.208102
**;TimeLapse Begin
G1 F9000 X0 Y190 ;Park print head
M400 ;Wait for moves to finish
M240 ;Snap Photo
G4 P700 ;Wait for camera
G0 X103.5 Y103.5; Restore position** 
;LAYER:1
M117 2/125 | ET 25M
M118 2/125 | ET 25M
M106 S25.5
;TYPE:WALL-INNER
;MESH:X-Y Ref Block.stl
          _G1 F2100 E70.27162_

So it doesn't really look like the same plugin.
Load a model, set Cura up to slice, use the "File | Save Project" command to create a project file.  Zip the 3mf file and post the Zip folder here.

darrabito, 2023-01-14:
Here you go:
[cylinder.zip](https://github.com/Ultimaker/Cura/files/10418834/cylinder.zip)

The timelapse setup I use is moonraker-timelapse plugin and I have a predefined TIMELAPSE_TAKE_FRAME macro from that with retraction disabled. So the problem is the unretraction before the macro is inserted and before the layer change

darrabito, 2023-01-14:
Here is the result (not the same file, sorry, but has the same settings):
[image]
[image]

As you can see the oozing is only present on one of the two threaded rods. Notably the one that gets a new layer after the other rod and then travels away for a timelapse photo

GregValiant, 2023-01-14:
The file you attached was "exported" and is only the model.  I need a Project File that will include your settings, printer, and the model.  
Below is the code generated with the "Time Lapse" post processor included with Ultimaker Cura 5.2.1.  You can see that the six added lines are between "TIME_ELAPSED" and ";LAYER:".  In you snippet all I see is a single line "TIMELAPSE_TAKE_FRAME".  I think we're talking apples and oranges here.  Are you using Ultimaker Cura or some other fork of the program?

G1 F4500 X141.677 Y111.184 E199.92858
G1 F2100 E193.92858
;MESH:NONMESH
G0 F600 X141.677 Y111.184 Z0.6
G0 F9000 X129.453 Y111.719
G0 X85.818 Y118.005
G0 X67.448 Y120.001
;TIME_ELAPSED:110.203192
**;TimeLapse Begin
G1 F9000 X0 Y190 ;Park print head
M400 ;Wait for moves to finish
M240 ;Snap Photo
G4 P700 ;Wait for camera
G0 X67.448 Y120.001; Restore position** 
;LAYER:2
M117 3/100 | ET 20M
M118 3/100 | ET 20M
M106 S51
;TYPE:WALL-INNER
;MESH:cylinder.3mf
G1 F2100 E199.92858

darrabito, 2023-01-15:
My bad, I didn't read. Here is the project file
[MyProject.zip](https://github.com/Ultimaker/Cura/files/10419151/MyProject.zip)

"TIMELAPSE_TAKE_FRAME" is simply a macro to handle all timelapse moves and camera controls provided by moonraker-timelapse. I am using MainsailOS+klipper and it integrates nicely. I am using a postprocessing script to insert "TIMELAPSE_TAKE_FRAME" at layer changes.

darrabito, 2023-01-15:
If you would like to inspect the timelapse configuration here is the file. But this is not where the problem lies as there are no retractions. Although I can add retractions in the macro, but I would like to keep all of those settings within cura.
[timelapse_macros.zip](https://github.com/Ultimaker/Cura/files/10419160/timelapse_macros.zip)

GregValiant, 2023-01-15:
You could use Search and Replace instead.
Search = ;MESH:NONMESH
Replace= TIMELAPSE_TAKE_FRAME\n;MESH:NONMESH
Use Regular Expressions = TRUE

That would give you this:
G1 F2400 **E43.73747**
**TIMELAPSE_TAKE_FRAME**
;MESH:NONMESH
G0 F300 X102.867 Y124.884 Z0.3
G0 F6000 X101.976 Y122.923
G0 X101.669 Y121.751
G0 X101.402 Y119.987
G1 F2400 **E44.53747**
G0 F6000 X101.752 Y120
G0 F6000 X101.752 Y120
;TIME_ELAPSED:58.593672
;LAYER:1
**G1 F2400 E43.73747**
M106 S255
;TYPE:WALL-INNER
;MESH:Cylinder
**G1 F2400 E44.53747**

That puts your "TIMELAPSE_TAKE_FRAME" between a retraction and a prime and takes "retract at layer change" out of the loop.  That should work so long as you don't change your travel/combing settings.

I don't see a bug here.  You have a third party printer customized by a fourth party and using fifth party firmware.  This is the way Ultimaker Cura works so this is expected behavior.

darrabito, 2023-01-15:
Okay, thank you for the information. I will switch to using search and replace

closed 2023-01-15 by darrabito