Ultimaker/Cura #14295: Build issue

SainathThota, 2023-01-14:
### Application Version

main

### Platform

windows 10

### Printer

code build issue

### Reproduction steps

Even i delete .conan folder i always get this error 

----Running------
> "sip-build"
-----------------
sip-build: An internal error occurred...
Traceback (most recent call last):
  File "C:\Users\thota\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\thota\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\thota\AppData\Local\Programs\Python\Python310\Scripts\sip-build.exe\__main__.py", line 7, in <module>
  File "C:\Users\thota\AppData\Local\Programs\Python\Python310\lib\site-packages\sipbuild\tools\build.py", line 37, in main
    handle_exception(e)
  File "C:\Users\thota\AppData\Local\Programs\Python\Python310\lib\site-packages\sipbuild\exceptions.py", line 81, in handle_exception
    raise e
  File "C:\Users\thota\AppData\Local\Programs\Python\Python310\lib\site-packages\sipbuild\tools\build.py", line 32, in main
    project = AbstractProject.bootstrap('build',
  File "C:\Users\thota\AppData\Local\Programs\Python\Python310\lib\site-packages\sipbuild\abstract_project.py", line 87, in bootstrap
    project.setup(pyproject, tool, tool_description)
  File "C:\Users\thota\AppData\Local\Programs\Python\Python310\lib\site-packages\sipbuild\project.py", line 595, in setup
    os.mkdir(self.build_dir)
FileExistsError: [WinError 183] Cannot create a file when that file already exists: 'C:\\.conan\\7aa46f\\1\\build\\Release'
ERROR: pyarcus/5.2.2: Error in generate() method, line 94
        sip.build()
while calling 'build', line 33
        self._conanfile.run(cmd, cwd = self._conanfile.source_folder)
        ConanException: Error 1 while executing "sip-build"

### Actual results

----Running------
> "sip-build"
-----------------
sip-build: An internal error occurred...
Traceback (most recent call last):
  File "C:\Users\thota\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\thota\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\thota\AppData\Local\Programs\Python\Python310\Scripts\sip-build.exe\__main__.py", line 7, in <module>
  File "C:\Users\thota\AppData\Local\Programs\Python\Python310\lib\site-packages\sipbuild\tools\build.py", line 37, in main
    handle_exception(e)
  File "C:\Users\thota\AppData\Local\Programs\Python\Python310\lib\site-packages\sipbuild\exceptions.py", line 81, in handle_exception
    raise e
  File "C:\Users\thota\AppData\Local\Programs\Python\Python310\lib\site-packages\sipbuild\tools\build.py", line 32, in main
    project = AbstractProject.bootstrap('build',
  File "C:\Users\thota\AppData\Local\Programs\Python\Python310\lib\site-packages\sipbuild\abstract_project.py", line 87, in bootstrap
    project.setup(pyproject, tool, tool_description)
  File "C:\Users\thota\AppData\Local\Programs\Python\Python310\lib\site-packages\sipbuild\project.py", line 595, in setup
    os.mkdir(self.build_dir)
FileExistsError: [WinError 183] Cannot create a file when that file already exists: 'C:\\.conan\\7aa46f\\1\\build\\Release'
ERROR: pyarcus/5.2.2: Error in generate() method, line 94
        sip.build()
while calling 'build', line 33
        self._conanfile.run(cmd, cwd = self._conanfile.source_folder)
        ConanException: Error 1 while executing "sip-build"
[image]

### Expected results

build success

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

MariMakes, 2023-01-20:
Hey @SainathThota,

😖 Sorry to hear you are stuck.

I'm only a fledgling member of the Build From Source Club 👑 
Before we get started, I have a few questions to narrow down the troubleshooting:

1. Can you make sure you have the latest Conan version?  
`pip install conan --upgrade `
You can check the [Conan Documentation]( https://docs.conan.io/en/latest/installation.html ) for the latest version and installation instructions

2. Did you try clearing the Conan Cache? 
`conan remove "*" -f `

3. Can you make sure you have the latest config with this command?
`conan config install https://github.com/Ultimaker/conan-config.git`

4. Can you make sure you have a default profile?
`conan profile new default --detect --force`

5. Can you make sure you have the cura-ce Artifactory repo and not the cura Artifactory repo?
Please check `cura remote list`, this will return your Artifactory Repo
If you see cura in your list, run ` conan remote remove cura`

6. With these steps you should have a clean slate and you can follow the steps on our wiki again. 
[You can find it here. ](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source) 

If you are stuck in any way, please reach out. We are happy to help out. 
The best way to do that is to provide us with your complete run-out log. 
Not just the part that fails, but all the lines generated after you've given your command. 
We need it see your profiles and packages. 

This is a neat trick to keep our Github issues more manageable. 
[image]

Can you help me understand. What do you want to achieve by creating a Build?
Are you intending to share it with others or are you using it for research purposes?
I would love to know more about your project, maybe we can help.

BluePotter, 2023-01-21:
I'm having the same problem as [SainathThot](https://github.com/SainathThota).  When I follow the steps from yesterday, when I got to step 5 cura was listed, so I removed it.  when trying to build again, conan cannot find pyarcus.
```
pyarcus/5.2.2: Not found in local cache, looking in remotes...
pyarcus/5.2.2: Trying with 'cura-ce'...
pyarcus/5.2.2: Trying with 'conan-center'...
ERROR: Failed requirement 'pyarcus/5.2.2' from 'conanfile.py (cura/5.3.0-alpha)'
ERROR: Unable to find 'pyarcus/5.2.2' in remotes
```
When I went through the steps again the I find cura had been restored to the list of conan remotes.  Not removing it and continuing, I get the original issue again.

BluePotter, 2023-01-22:
[build.with.cura.txt](https://github.com/Ultimaker/Cura/files/10475368/build.with.cura.txt)

astead, 2023-01-25:
I ran into the same issue as [SainathThot](https://github.com/SainathThota).
Trying to clear the conan cache and got the same thing again.

wasmann, 2023-01-31:
I got the same error too

BluePotter, 2023-02-01:
I just noticed an anomaly in my stack trace of the final error.
`sip-build: An internal error occurred...
Traceback (most recent call last):
  File "C:\msys64\mingw64\bin\sip-build-script.py", line 33, in <module>
    sys.exit(load_entry_point('sip==6.7.5', 'console_scripts', 'sip-build')())
  File "C:\msys64\mingw64\lib\python3.10\site-packages\sipbuild\tools\build.py", line 37, in main
    handle_exception(e)
  File "C:\msys64\mingw64\lib\python3.10\site-packages\sipbuild\exceptions.py", line 81, in handle_exception
    raise e
  File "C:\msys64\mingw64\lib\python3.10\site-packages\sipbuild\tools\build.py", line 32, in main
    project = AbstractProject.bootstrap('build',
  File "C:\msys64\mingw64\lib\python3.10\site-packages\sipbuild\abstract_project.py", line 87, in bootstrap
    project.setup(pyproject, tool, tool_description)
  File "C:\msys64\mingw64\lib\python3.10\site-packages\sipbuild\project.py", line 595, in setup
    os.mkdir(self.build_dir)
FileExistsError: [WinError 183] Cannot create a file when that file already exists: 'C:\\.conan\\90223d\\1\\build\\Release'
ERROR: pyarcus/5.2.2: Error in generate() method, line 94
        sip.build()
while calling 'build', line 33
        self._conanfile.run(cmd, cwd = self._conanfile.source_folder)
        ConanException: Error 1 while executing "sip-build"`
The key being `C:\msys64\mingw64\lib\python3.10\`.  I hadn't setup type python in msys, I had setup the python in AppData.  Making adjustments to my path, and repeating using Python3.10 in AppData I still get the same error, but a more sane stack trace.

[session.txt](https://github.com/Ultimaker/Cura/files/10552849/session.txt)

BluePotter, 2023-02-01:
Minor change, for others not versed in pip, the first command is:
`pip install conan==1.56.0 -force`

BluePotter, 2023-02-02:
That fixed it for me.  It now builds to completion.

closed 2023-02-03 by github-actions[bot]

TravisRo, 2023-02-12:
`pip install conan==1.56.0 -force`
This fixed it for me as well.  Seems like someone should either update the build instructions or fix the scripts so they work with the newest version of conan.

On a separate note, I also had to `pip install sip` which is also missing from the install instructions.

akhiljacob-trenser, 2023-09-21:
> e lis

I faced a similar issue while building Cura 5.4.0.
As advised I followed these procedures as mentioned below
* pip install conan==1.56.0 -force
* pip install sip (I even tried)
 Still, I am getting the same issue with the pyarcus sip.build()

[image]