Ultimaker/Cura #14296: Print overhangs starting from existing edges

Timmmm, 2023-01-14:
### Is your feature request related to a problem?

I want to print a shape like this (cross section):

[image]

Cura slices it like this:

[image]

And it actually prints the inner wall first, with zero support.

[image]

Clearly this will not print, but I think that it could!

### Describe the solution you'd like

There should be an option (or maybe just always do this): `Print overhangs starting from existing edges`. Basically the algorithm would be something like this (very roughly)

```
for each layer:
   find all the overhangs and supported parts of the layer
   print supported walls and infill first
   re-order remaining lines such that the ones that are best supported are printed first (i.e. shortest and highest percentage of touching edge)
```

I think that should be fairly achievable and give vaguely good results.

A perfectly optimal solution would be to print in a spiral from the outside in, but that may be asking too much (a general version of #14036 that wasn't restricted to circles could probably do this.   

### Describe alternatives you've considered

Adding supports :-/

### Affected users and/or printers

People who print overhangs.

meltiseugen, 2023-01-15:
Another way would be to add an entire full bottom layer (bridge) to close the hole, and when the print is finished, you could just break that layer.

Timmmm, 2023-01-15:
I have tried that actually (with PrusaSlicer rather than Cura) - basically a cylinder under the unsupported inside edge.

Maker's Muse used this technique and someone suggested it to me on StackExchange.

It worked way better than tree supports, and was very easy to remove. Unfortunately it still doesn't quite work because bridging doesn't work when one of the ends is very thin apparently.

So the bridges that go all the way across without touching the inner circle work fine, but the ones that go from the outer circle to the inner circle don't stick to the inner circle, so you just get loops hanging down.

Here's a blurry photo:

[image]

So I think it would still work better if it spirals in from the outside. Is there a way to hand edit a particular bit of GCode somehow? Don't think I'll have time to actually compile Cura or PrusaSlicer and implement any of this but maybe if I can show it works someone else (without children!) can.

Otherwise I'll probably just redesign my part to avoid 90 degree overhangs.