Ultimaker/Cura #14301: Failed to install any plugins in Cura 5.2.1 on MacOS 13.1

froeschel2001, 2023-01-15:
### Application Version

5.2.1

### Platform

MacOS 13.1 (M1-Chip)

### Printer

-

### Reproduction steps

1. Installed Cura
2. Loaded Plugin over Marketplace (loads a few minutes then aborts, instal button reappears)
3. Added Plugin over web
4. Restarted Cura - Sync changes
5. After a few minutes error appears: 1 plugins failed to download

### Actual results

No plugins can be installed on my Mac System

### Expected results

Plugins should be able to be installed on MacOS

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10421074/cura.log)

MariMakes, 2023-01-20:
Hey @froeschel2001,

Thanks for your report 👍

I think I'm reading that you are having trouble downloading the Settings Guide Plugin.
Youc an also download it here: https://github.com/Ghostkeeper/SettingsGuide/releases/tag/v2.9.2
and than drag it into Cura.

`ERROR - [MainThread] Marketplace.PackageList._downloadError [251]: Failed to download package: SettingsGuide2 due to <?xml version='1.0' encoding='UTF-8'?><Error><Code>ExpiredToken</Code><Message>The provided token has expired.</Message><Details>Request signature expired at: 2023-01-15T19:52:51+00:00</Details></Error>
`
Could it be that you've had your Cura open for a long time without internet connection? 
Did you try signing in and out again or rebooting Cura? 

I hope this helps you get unstuck 😇

MariMakes, 2023-01-20:
Hey @froeschel2001,

Glad to hear that it is resolved 😊 
There is a token that needs to be updated regularly, perhaps the timing was just unfortunate 

Good luck with learning about the Cura settings 💪 
I'll close the issue

closed 2023-01-20 by MariMakes