Ultimaker/Cura #14302: 5.3.0-alpha+xmas unneeded supports

pantala, 2023-01-15:
### Application Version

5.3.0-alpha+xmas

### Platform

MacOS M1

### Printer

Prusa i3 MK3S+

### Reproduction steps

1. Load STL (attached project file)
2. Slice using new xmas tree supports

### Actual results

Unneeded supports generated in the middle of the model where there is no overhang.

### Expected results

Supports generated only on the outer rim where there is an overhang. (This doesn't happen with normal supports.)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[PI3MK3M_Wheels_frontrim.3mf.zip](https://github.com/Ultimaker/Cura/files/10421205/PI3MK3M_Wheels_frontrim.3mf.zip)
[image]
[image]

DerGenaue, 2023-01-15:
This is probably the same as has been reported in this discussion: https://github.com/Ultimaker/Cura/discussions/14111?sort=top#discussioncomment-4585873
Te bug has already been fixed for the next version

DerGenaue, 2023-01-15:
If you want to, you can try out this fix by the original contributor of the new tree support algorithm:
Be aware that this is not an official Ultimaker build though and that you need to be on Windows to test it
https://github.com/ThomasRahm/CuraEngine/releases/tag/5.3.0-alpha%2Bxmas%2Bfixes

MariMakes, 2023-01-20:
Hey @pantala,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for the analysis @DerGenaue 🤩 

We've also received a report on this on our Community Forum.
And have already added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10179

Thanks for the report! 👍

DerGenaue, 2023-07-04:
Should be solved in 5.4

MariMakes, 2023-07-11:
True 😊 
I'm closing this issue.

closed 2023-07-11 by MariMakes