Ultimaker/Cura #14304: Support blockers

OldBlood122, 2023-01-15:
### Application Version

5.2.1

### Platform

win 11

### Printer

E3 S1 Pro

### Reproduction steps

Added support blocker to the corner of a model because for some reason, it was adding supports there when there was no reason for them to be there. It was a plain vertical wall.

### Actual results

Nothing, it didn't block supports.

### Expected results

Supports blocked with support blocker cube.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

https://www.printables.com/model/277194-quick-release-camera-mount-ender-3-s1-pro

This is the file i am working with

GregValiant, 2023-01-15:
Thanks for the report.
I'm not sure which of those models you were slicing.  Without a project file we have no idea of how you set the slice up or how you configured the support blocker.  I can't duplicate this on my own as I've had no trouble with the blockers.
Use the "File | Save Project" command to create a project file.  Zip the file and post the zip folder here.  Someone will take a look.

OldBlood122, 2023-01-15:
Ok sorry about that wasn't sure how to do that. I'll do it soon.
Thanks you.

Asterchades, 2023-01-16:
Make sure that you're placing the blockers against where the supports are going _to_, rather than where you don't want them to pass through. While the new trees in the 5.3 alpha will avoid blockers entirely, regular supports and the existing supports use them to do a simple check - any surface within the confines of a blocker will not be supported, but anything else (including areas _above_ a blocker) is fair game.

If that doesn't help, be aware that you cannot upload files via the e-mail response system. You'll need to upload your project via the website proper.