Ultimaker/Cura #14305: Old cura profiles that no longer exist are coming back to haunt me

GoguyT3D, 2023-01-16:
### Application Version

5.0

### Platform

Windows 11

### Printer

Ultimaker S3

### Reproduction steps

Every time something is printed a 3mf is saved to "recall" or "reprint" a model, saved to my HDD. I use third party filaments which frequently need to be updated with improvements to the profiles for better print quality. Each time a profile is updated I would expect cura to automatically use the "updated" profile when an old 3mf file is loaded although it does not. Some 3mf files which were saved a very long time ago when I was using different profiles with different names (.2mm TPU) and the profiles have since been removed or renamed for compatibility within cura, the old profile POPS into cura and sticks into my list of profile names. Within cura when clicking on the old profile to remove it a crash occurs. How do I wipe these unwanted profiles permanently?

### Actual results

An old profile which was thought to be removed is magically re added to my profile list probably from the 3mf loading, even though I though I imported all models from old 3mfs and resaved them with the new profiles, maybe I missed one but why does this need to be such a hassle, and why do the old profiles not update to the new ones?

### Expected results

3mfs should automatically select active profiles that exist within cura so Old profiles that create crashes dont get pasted in

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[image]
The Profile labeled ".2mm TPU" is very old not used and was removed when I received the "your files must be corrupt" crash that required me to completely remove all profiles reload all profiles and add all my printers from scratch agian.
[cura.log.zip](https://github.com/Ultimaker/Cura/files/10421771/cura.log.zip)
The original Cura.log.1 file isn't supported so I zipped it

GoguyT3D, 2023-01-16:
[machine_instances.zip](https://github.com/Ultimaker/Cura/files/10421796/machine_instances.zip)
Config files added also

GoguyT3D, 2023-01-16:
[quality_changes.zip](https://github.com/Ultimaker/Cura/files/10421800/quality_changes.zip)
Looks important

GoguyT3D, 2023-01-21:
@MariMakes maybe you can help me understand how to either wipe all old profiles from the file folder and re-load my updated ones or something maybe a little easier. Thank you

MariMakes, 2023-01-26:
Hey @GoguyT3D,

Good to see you again 😄 
Sorry, it took us a while to get back to you 😞 [It was a crazy week](https://ultimaker.com/3d-printers/ultimaker-s7).

I'm not totally sure if this workflow is supported. But we want to start improving the profile workflows, so we should definitely take this into account. So I have three options for you. 

1. Have you considered loading your old project files, just the models?
[image]
You will keep the orientation, but the settings will match your new profile.
It will result in having to enable/disable support and adhesion but that seems less work than removing profiles. 

2. A project file is secretly a .zip with a bunch of definitions. 
[image]
If you are consistent you can remove the mentions of your profile in at least 4 places, and you will still be able to load your project file without loading the profile. But that seems more like a hack and you are risking corrupting part of your Cura. 

3. Would you like me to ask the team if we can make a feature request to add "Ignore Profile" here? 
[image]

What applies best to your situation?

GoguyT3D, 2023-01-26:
It's been crazy for all of us no hard feelings, I would like to update you, I  have tried option #(1) with all of my saved 3mfs importing the model only with the appropriate updated profile selected and resaved them all, this was after cura said "your files must be corrupt " , it took a long time as I have hundreds of old 3mfs , I feel like option #(2) would be too risky since I've already been slapped with the corrupt files situation so I'll say I'd like option #3 if cura allowed you to select a file to open, brought up the upload summary,*** asked if you wanted to use the unsupported (or old) profile or import the model to use a current profile by my selection *** that would be nice,  thank you for responding

MariMakes, 2023-01-26:
Thanks @GoguyT3D,

Wow, that sounds like a lot of work. 😅 
I'll bring number 3 up to the team. 👍 
Fingers crossed 🤞

GoguyT3D, 2023-01-26:
@MariMakes, I think I remember what 3mf file was to blame for this madness. In the meantime (to remove the unwanted profile) what might you recommend I do. Every time I click on the profile in cura (to then click remove) it crashes cura. Is there a workaround in the configuration files to remove it or maybe from the cashe memory?

GoguyT3D, 2023-01-28:
Stupid me I forgot about cura backups, Ignore the last question.

MariMakes, 2023-02-09:
We've added a ticket to the backlog with the intent to improve this. 
We are going to try to include an ignore profile option. 

For internal reference CURA-10258

Thanks for the report! 👍