Ultimaker/Cura #14307: Tree support through part

mslucass, 2023-01-16:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ender3 with BTT SKR

### Reproduction steps

slice the part and the support will go through the part instead of around it

### Actual results

[image]

support through part

### Expected results

only support on the build plate next to the hole

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3E3V2_051 - 24v cover.zip](https://github.com/Ultimaker/Cura/files/10428621/CE3E3V2_051.-.24v.cover.zip)

GregValiant, 2023-01-16:
Thanks for the report and the project file.
I can duplicate this with your project.  The Cura team will take a look.

For this model I wouldn't bother to support those holes.  They are small and should bridge with no issues.  That would allow you to use normal supports and "touching buildplate" which will eliminate the "Going Through The Model Problem".

DerGenaue, 2023-01-16:
Please try the new [Tree Support XMas Alpha](https://github.com/Ultimaker/Cura/discussions/14111).
It should solve this problem

closed 2023-01-30 by mslucass