Ultimaker/Cura #14311: Starts slice put won't slice

rlebkuecher, 2023-01-17:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Creality Ender 5 Plus

### Reproduction steps

Tried to slice a new object and it failed. I resaved the object from Tinkercad as a stl again with the same results.
[image]

### Actual results

The Error message popped up saying Slicing Failed with an unexpected error.

### Expected results

I expected the item to slice.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10435413/cura.log)
[Phone Holder v6.zip](https://github.com/Ultimaker/Cura/files/10435440/Phone.Holder.v6.zip)

rlebkuecher, 2023-01-17:
In Tinkercad I undid the last grouping of the part then regrouped it without any changes to the design. After that I exported the object again as an stl and it sliced just fine.

GregValiant, 2023-01-17:
Thanks for the report.
I didn't find anything wrong with the model you posted and it sliced fine for me.
It seems to have been a model error rather than a Cura bug.  Can we close this then?

closed 2023-01-17 by rlebkuecher