Ultimaker/Cura #14312: Slicing a large model freezes around halfway

fifex-ltd, 2023-01-17:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Creality CR10 

### Reproduction steps

Loaded a large model into Cura for slicing , slicing hangs around halfway.
I've left it for around 30 + minutes and it doesn't progress.
I'm using a Dell Precision 3541 workstation 32gb of RAM and i7 2.6ghz to slice this.
Haven't had any othe rproblems with smaller models.
I've included the project file , log and a screenshot of the behaviour.

### Actual results

Slicing hangs around halfway 

### Expected results

Slicing should complete

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10437143/cura.log)
[image]
[CCR10MAX_gogglers case base part.zip](https://github.com/Ultimaker/Cura/files/10437204/CCR10MAX_gogglers.case.base.part.zip)

GregValiant, 2023-01-17:
Thanks for the report and the file.
I left the house with this still slicing so I have no idea exactly how long this took to slice but it took a while.  I had moved it and rotated it in Cura.  Someone from the Cura team will take a look.

[image]

fifex-ltd, 2023-01-18:
Yes I had the same thing , left it running on my desk and when I came back several hours later it was done.
I'll try and do it again today and see if I can record how long it takes to complete.