Ultimaker/Cura #14314: Selecting "change settings visibility" dropdown has no effect

VincentJGeisler, 2023-01-17:
### Application Version

5.2.1

### Platform

macOS ventura

### Printer

ender 6s plus

### Reproduction steps

click 3 bars and set to "custom selection"

go to the section you want to change setting visibility on, click the "bubbly 3 bars" 

Nothing happens. 

### Actual results

There is no expansion of which settings you want to become visible

### Expected results

an expansion of the items that are available to select for display under custom selection. 

I have to select "all" in order to change the options I need, which is annoying as heck. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

ZxZX

VincentJGeisler, 2023-01-17:
After some fiddling around, I discovered that the popup window for settings visibility is being popped up BEHIND the main window, hiding it from view. 

While the solution is to minimize the main window and bring the settings visibility window to the forefront, this is a bug as the user expects a pop up to occur in front of the main window not behind.

MariMakes, 2023-04-05:
Hey @VincentJGeisler,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you, it seems like your issue must have gotten lost in our pile😞 

To me, it doesn't seem like a bug in Ultimaker Cura, since there is a solution in place. 
Apologies for any inconvenience caused by our delayed response. 
I'll be closing this issue. I hope you understand.

closed 2023-04-05 by MariMakes