Ultimaker/Cura #14317: cant find printer

raz912, 2023-01-18:
### Application Version

5.2

### Platform

windows 10

### Printer

creasee cs-30

### Reproduction steps

plug and unpluged printer even restarted computer with it pluged in

### Actual results

didnt find

### Expected results

find so can print from computer

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

non to add

GregValiant, 2023-01-18:
Thanks for the report.
USB printing is included with UltiMaker Cura as a legacy plugin.  Because no UltiMaker printers use it, the UltiMaker company no longer supports it.  It either works or it doesn't.  There are just so many hardware/software/driver combinations out there that supporting USB printing is nearly impossible.

Since you are on Windows you can plug the printer in to a USB port and check "Device Manager" and see if the computer is actually seeing the printer.  Under "Ports COM and LPT" you should see something like "USB-SERIAL CH340(COMx)" where "x" is the com port number.  If it isn't there then maybe you don't have the correct port driver installed for your printer?  
An option would be to slice with Cura and pass the gcode to Printrun/Pronterface (or some other print server) and see if you can get that to work for you.

MariMakes, 2023-01-20:
Hey @raz912,

Welcome to the Ultimaker Cura Github 🚀 

@GregValiant is right, there are A LOT of printer software combinations, and it's hard to keep track of them. 
Perhaps you can find a dedicated Creasee forum to ask your question? 

Or perhaps there is a problem with your USB Printing Plug-in. Can you doublecheck if it has been enabled? 
You can find it in the Marketplace after you click on the Gear. 
[image]

closed 2023-02-03 by github-actions[bot]