Ultimaker/Cura #14319: Slicer Crashing

FenGryn, 2023-01-18:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

I tried to slice an articulating dragon, the slicer crashed, over, and over, and over, and over again

### Actual results

Literally nothing useful, an unknown error and a crash, every single time I try

### Expected results

A successful gcode should be produced

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I have nothing more to add and find it annoying you demand I do

FenGryn, 2023-01-18:
Surprisingly, not only was PrusaSlicer able to slice it without crashing, but with quicker times.

GregValiant, 2023-01-18:
Thanks for the report.
_"...I have nothing more to add and find it annoying you demand I do."_
I'm sure "asking" for support files was not a demand.  Please understand that without a Project File and Cura Log there can be no investigation.  We have no idea what might be going on.  I know that a model called a "Crystal Dragon" had hundreds of errors in it.  There was another articulated dragon that looked to be made of feathers that had a lot of errors as well.  This could be a Cura problem, a settings problem, or could be a model problem.  As I recall PrusaSlicer has an auto-repair function where Cura's is manual and the Mesh Tools needs to be loaded from the Marketplace in order to use it.
Github requires that any files other than ".txt" files need to be zipped in order to be posted.  That isn't an UltiMaker rule but a rule by the website people.

closed 2023-02-01 by github-actions[bot]