Ultimaker/Cura #14321: Rehome X Y between layers

jimkazmer, 2023-01-18:
### Is your feature request related to a problem?

On large/long prints the X and/or Y print head location(s) can get shifted (for many possible reasons).  These offsets tend to accumulate, sometimes they average out, sometimes they grow to the point that the print is considered failed. 

### Describe the solution you'd like

I'd like the option to tell Cura to re-home X and Y between layers. This will cost time, but it should minimize/avoid the cumulative effect of layer shifts.  This could be done without printer firmware changes.

When re-homing, it would be useful if the some metrics of the degree the printer was off was saved/reported.  For example, if the printer re-homes between layer 23 and 24, and finds that one dimension was off by more that N mm, then the user might want to cancel the print. I know for that capability, the printer firmware would need to get involved.

### Describe alternatives you've considered

I have watched so many videos on the reasons for layer shifts, and have followed the advice. Yet, they still happen sometimes.

### Affected users and/or printers

Everyone who has a printer that sometimes has layer shifts.

GregValiant, 2023-01-18:
This is an interesting one.
I have never gotten any "drift" in a print.  That could be because of the smaller size of my printer (230 x 230).
I don't think it would be possible to inform a user of any drift.  The printer wouldn't know as all it knows about location is the "step count".  The printer will display XYZ location but that is a guess given the number of steps sent.

The end stop switches are not exactly repeatable.  There will be some change in the absolute "0" of the X, Y, or both.  I know I have issues with my Z switch and the other two are the same design.

You can accomplish what you want by using the "Insert at Layer Change" plugin.  You would insert "G28 X Y".
If you were to decide that is too often then it gets more difficult and you would probably need multiple instances of Search and Replace.

jimkazmer, 2023-01-18:
Thank you for the comment and info.  I will give the "Insert at Layer Change" a try and add the results here.

I am printing on a 400x400mm bed; the print is covering 65% of that space with the model going corner to corner along one diagonal, and the model is 28.5mm high.

For the "G28 X Y" to work well, I realized I need to recompile my marlin firmware to perform the re-zeroing more slowly. I have one printer that re-zero's the X & Y very slowly after finding the limit switch, and my large "problem" printer re-zero's so fast that it is likely highly variable.

The printer (firmware) should know its (previous) offsets as it re-zero's... When it hits its limit switch, it should know something like, "huh, I though I was at 1.235mm".  Yet that is a firmware feature not a slicing feature.

GregValiant, 2023-01-19:
After thinking about this a bit I altered the "InsertAtLayerChange" plugin.
I've added the option for "every so many layers" and multi-line support for the commands.  Seems to work fine.

If you want to be a guinea pig put this in the (Windows):
"C:\Program Files\Ultimaker Cura 5.2.1\share\cura\plugins\PostProcessingPlugin\scripts" folder.  It will show up in addition to the stock plugin as "Greg's Insert at Layer Change".

[GVInsertAtLayerChange.zip](https://github.com/Ultimaker/Cura/files/10452990/GVInsertAtLayerChange.zip)

[image]

GregValiant, 2023-01-21:
I played around some more.  There are new options to start at the layer of your choice and end at a layer of your choice.

jimkazmer, 2023-01-21:
Wow, that is awesome.
I will give it a try.  My layer shifts were due to a very slight over-extrusion that manifest after layer 40-something and 0.5kg of filament.  It would be cool if I could somehow log the position delta... I guess I could enhance Marlin to save the delta's to the eeprom whenever it homes.

I plan to test your new feature sometime in the next week.  Thank you for making it possible.

GregValiant, 2023-01-21:
When you query the printer for the current position the command is M114 (provided it's enabled in your firmware).  The printer would respond by sending the current location across the USB.  You would need to catch the response from the printer with a Print server app (Pronterface/Printrun would work.  Below is a download link to my own app.)
The response would be something like:
X:115.00 Y:115.00 Z:5.00 E:0.00 Count X:9200 Y:9600 Z:2000

A CNC machine would typically have a scale that the processor would read to insure the machine is where it thinks it is.  There is positive feedback.  Our printers don't have that as it would be expensive to implement.  A consequence is that the printer doesn't really know where the nozzle is.  It knows it's own "Steps/mm" for an axis and interprets an X Y Z E position based on the steps/mm.  Consider if you removed the X belt...the printer would still report the position as X115 even though the nozzle never moved in the X.
In the line above you can see that although the location is given as X115 Y115 the step count is different for the X and Y.  In both cases the Steps/mm is 80.  The Y has traveled an additional 5mm because my Y Home Offset is -5 while my Home Offset X is 0.  So the reported "Steps" are from the Auto-Home position rather than from the Home Offset 0,0,0 while the reported "location" is indeed from the Home Offset 0,0,0.

You didn't mention your operating system.  I wrote the app below to communicate with the printer.  It's for Windows only and was written with MarlinFW in mind.  If you choose to install it you would probably need to fool your anti-virus as it doesn't have a digital signature.  There is a readme and some instructions (such as they are).  There is a desktop icon and an un-install app.  If you decide to install it I'd appreciate any feedback and/or bug reports.  You can message me here or on the UM Forum where I have the same username.

[Greg's Toolbox](https://drive.google.com/uc?export=download&id=1rntbmjRc4MLaDo7F2DbN2BwZWjBAJ8iJ)

jimkazmer, 2023-01-23:
I understand what you are saying.  I plan to collect the data from within a modified version of Marlin.   I will modify Marlin to record some data every time it Homes.  The first time it homes (after power-on), it has not idea where the print head is, yet it should know "this is the first time I'm homing after a power-on".  After that, my modified Marlin firmware will record the positional X,Y,Z before it resets each value to zero. I'll need to learn a bit more to get all the details, and develop the data extraction mechanisms.

GregValiant, 2023-01-23:
I post processed a gcode file to include an M114 at every layer change.  The data recorded on the text box in the Toolbox app.  I copied it over to Excel and went through it with a macro.  Everything was dead on as I had expected.  I suspect that without a second way to measure the displacement on the X and Y that you will come across the same thing.  The printers just aren't very bright.

jimkazmer, 2023-01-23:
Did you try forcing a dislocation (missed steps)?

GregValiant, 2023-01-23:
If you install that app, one of the tools is the Acceleration and Jerk tool.  It's a series of different "courses" around the bed.  You can change the Speed, Accel, and Jerk settings to see how your printer responds.  You can certainly jack them up and cause missed steps.  The way I've written it though - each course starts and ends with an Auto-Home and although I know I've cause missed steps, I've never checked for movement accuracy.  I assume it lost it's way.   There certainly would have been layer shifts.

The extruder would probably be the easiest to check.  Hold the shaft with a pair of pliers and cause missed steps.  The shaft wouldn't move.  I wonder what the E location would be?