Ultimaker/Cura #14324: Sliced objects wrong display.

x-channel, 2023-01-18:
### Application Version

5.2.1

### Platform

Ubuntu 22.04.01 LTS

### Printer

Creality Ender 3 pro

### Reproduction steps

Install RX 6600 from Gigabyte.
Install Ubuntu 22.04.01
Install proprietary driver for AMD.
Run Cura.
Slice Object.
Enter in Preview mode.
Move camera.

### Actual results

Behold behind sliced object.

### Expected results

See only outter object shell

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I suppose that happens due miss reference for one vertex from each quad face.

[image]

MariMakes, 2023-01-20:
Hey @x-channel,

Wow! That's a super spacey preview. 😵 
This issue is outside my field of expertise. 😅 

I'll bring it up to the team to see what they can do to improve it. But I expect that if it's not a quick fix that it will not get a high priority to implement. Fingers crossed 🤞

x-channel, 2023-01-20:
Thanks to reply @MariMakes!

x-channel, 2023-01-22:
@MariMakes small update.

I've reinstalled ubuntu. My machine running KUbuntu 20.04.5 and KUbuntu 22.04.1, both with opensource driver. Cura 5.0.0 running perfect in deprecated ubuntu, but keep bug in latest SO version. I will search more, but think that could be some crashed python lib, or some crashed ubuntu lib that called by python lib. I will test again in intel UHD 630 as soon as possible.

snoopen, 2023-01-31:
As I mentioned in issue #14343 this also affects me:

First slice it appears fine, but any interaction (to rotate or move model/view, layer slider, etc) all breaks the render. However top most layer (as changed with the slider) always appears fine, just all the other layers render incorrectly.

Tested with all Appimages 5.3.0-alpha back to 4.131 (modern where available)

ThinkPad T14 Gen 3
Pop!_OS 22.04 LTS x86_64
Gnome 42
OpenGL version string: 4.6 (Compatibility Profile) Mesa 22.2.0

[image]

snoopen, 2023-01-31:
I've tested a few other 3D programs and games and none that I tested appeared to have any rendering issue. So I'm not convinced this is a driver issue, not at the OS level at least. Perhaps a library issue that Cura relies on and bundles?