Ultimaker/Cura #14325: Cura crashes regularly on Macbook Pro M1

theneverafter, 2023-01-18:
### Application Version

5.2.1

### Platform

MacOS 12.4

### Printer

Creality Ender 5 Pro

### Reproduction steps

1. Started app
2. Sliced model / regular usage

### Actual results

App crashes randomly. 
[cura.log](https://github.com/Ultimaker/Cura/files/10448852/cura.log)
[Ultimaker-Cura-2023-01-13-175906.txt](https://github.com/Ultimaker/Cura/files/10448894/Ultimaker-Cura-2023-01-13-175906.txt)

### Expected results

App runs stable.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Logs added above.

yuval2508, 2023-01-19:
Same to me - MacBook Pro 16" Intel architecture. Crashes sporadically after 2-3 times using it to slice, clearing the build plate  after slicing extends the next crash by few more time.

MariMakes, 2023-04-05:
Hey @theneverafter,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you, your issue must have slipped to the bottom of the pile 😞 

The new Cura 5.3 Stable contains the fix for this for a number of slicing crashes. 
Could you test if that's the case for your model as well? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

theneverafter, 2023-05-05:
@MariMakes I just downloaded 5.3.1 and the crash issue is still happening. 

Here is a project file: [travel_bag_support_long.3mf.zip](https://github.com/Ultimaker/Cura/files/11408303/travel_bag_support_long.3mf.zip)

And here is the crash info: [Ultimaker-Cura-2023-05-05.txt](https://github.com/Ultimaker/Cura/files/11408372/Ultimaker-Cura-2023-05-05.txt)

yuval2508, 2023-05-05:
Same to me. It happens even if the software is just open without slicing anything. 
I can produce the crash report.

Piezoid, 2023-05-05:
@theneverafter Can you also upload your cura.log? On maxOS, it's located under this path `$USER/Library/Application Support/cura/5.3/cura.log`

The aborting thread is the `QEventDispatcherUNIX` loop. There is a `QMessageLogger::fatal()`  call on the stack, meaning that useful information about the crash might be present in the log:
```
Thread 130 Crashed:
0   ???                           	    0x7ff895c229a8 ???
1   libsystem_kernel.dylib        	    0x7ff80688522a __pthread_kill + 10
2   libsystem_pthread.dylib       	    0x7ff8068bcf7b pthread_kill + 263
3   libsystem_c.dylib             	    0x7ff8067c9480 raise + 26
4   libsystem_platform.dylib      	    0x7ff8068e7c1d _sigtramp + 29
5   ???                           	               0x0 ???
6   libsystem_c.dylib             	    0x7ff806806ca5 abort + 123
7   QtCore                        	       0x11005c559 0x110052000 + 42329
8   QtCore                        	       0x1103c6e4b QMessageLogger::fatal(char const*, ...) const + 213
9   QtCore                        	       0x1103d8e1e 0x110052000 + 3698206
10  QtCore                        	       0x11026e483 QEventDispatcherUNIXPrivate::QEventDispatcherUNIXPrivate() + 179
11  QtCore                        	       0x11026eed5 QEventDispatcherUNIX::QEventDispatcherUNIX(QObject*) + 37
12  QtCore                        	       0x110273c89 0x110052000 + 2235529
13  QtCore                        	       0x1101e9cde 0x110052000 + 1670366
14  QtCore                        	       0x110273e19 0x110052000 + 2235929
15  libsystem_pthread.dylib       	    0x7ff8068bd259 _pthread_start + 125
16  libsystem_pthread.dylib       	    0x7ff8068b8c7b thread_start + 15
```

MariMakes, 2023-05-08:
Hey All, 

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/15161
We have a ticket on our backlog with the intent to improve this behavior.

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/15161

closed 2023-05-08 by MariMakes