Ultimaker/Cura #14326: Slicing with .3 Initial Layer Height failed

trbaldwin60, 2023-01-18:
### Application Version

5.2.1

### Platform

Ubuntu Linux 20.04LTS

### Printer

Ender 3 V2 with Octopus IDEX upgrade

### Reproduction steps

RE: https://www.thingiverse.com/thing:2014307 Baby Groot

Set Initial Layer Height to .3 using PLA wood with .6 nozzle, and using Baby Groot lower detail model sitting (rotated 90 degrees for sitting).
Click Slice and it immediately fails.
However, if I leave Baby Groot on his back, it slices OK, but this is undesirable.
Also if I move the model forward just a little bit, and/or rearrange after rotation, it slices OK

I'm unable to upload the gzip project file because it is 43MB and your web site won't allow more than 25MB.

### Actual results

Slice fails

### Expected results

Slicer should work.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

trbaldwin60, 2023-01-18:
Also if I change the Initial layer height to .2 or .4, it slices. So the issue seems to only occur when set to .3.

MariMakes, 2023-06-05:
Hey @trbaldwin60,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

The new Cura 5.3 contains the fix for this for a number of slicing crashes. 
Could you test if that's the case for you as well? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-06-20 by github-actions[bot]