Ultimaker/Cura #14327: Unknown Slicing Error

harleyriderne1969, 2023-01-18:
### Application Version

5.3.0-alpha+xmas

### Platform

Windows 11

### Printer

Sovol SV05

### Reproduction steps

Trying to slice a lithphane and getting an error.  Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker.

### Actual results

Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker.

### Expected results

Successful slicing

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10451237/cura.log)
[SS_top_lithophane-box1.zip](https://github.com/Ultimaker/Cura/files/10451240/SS_top_lithophane-box1.zip)

GregValiant, 2023-01-19:
Thanks for the report and the files.
The model has a couple of errors.  I exported it from Cura and repaired it with MS 3D Builder and then it sliced.
There is a bug that can keep models from slicing when they are at the midpoint of the build plate in Cura.  You could try moving and/or rotating the model and it might fool the bug.
Here is the report on the model:
--> 4 Naked edges (?)
--> 1 Planar holes (?)
--> 0 Non-planar holes (?)
--> 0 Non-manifold edges (?)
--> 0 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 119585ms ------
-> Vertex count changed from 179359 to 181030 (+1671)
-> Triangle count changed from 358712 to 362056 (+3344)

I'll mark this as a duplicate slicing error for now.  If you get it to slice let us know.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker