Ultimaker/Cura #14328: Account Sync every 60s forces re-slice, errors in post process screen

Whitespace, 2023-01-19:
### Application Version

5.2.1

### Platform

macOS Ventura 13.1 (22C65)

### Printer

Ultimaker S5

### Reproduction steps

1. Log into your account in the upper right
2. Slice a model
3. Wait 60s for the account to automatically sync or click on **Check for account updates**

### Actual results

The slice preview is lost. Whatever you were doing you have to re-do it.

### Expected results

Nothing happens.  You don't lose your sliced preview when your account is synced.  Certainly not every 60s.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

None

MariMakes, 2023-01-20:
Hey @Whitespace,

Welcome to the Ultimaker Cura Github 🚀 
Wow! That sounds super annoying😞 
It's the first time I've heard of it. 

Do you mind sharing your logs? 
You can find your log file here:
MacOS: $USER/Library/Application Support/cura/<Cura version>/cura.log

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

Whitespace, 2023-01-20:
Here you go: [cura.log](https://github.com/Ultimaker/Cura/files/10469275/cura.log)

Just now I opened up Cura, opened a recent project (NB: that has several ChangeAtZ post-processing scripts in it), sliced, then clicked on "Check for account updates," and confirmed it's happening.  Also it happens on its own every minute as the account sync runs.

MariMakes, 2023-01-26:
Hey @Whitespace,

That's the file I was looking for, thanks for sharing 👍 
I'll bring it up to the team, hopefully we can find something fast 🤞