Ultimaker/Cura #14331: Multi material painting

knownchild, 2023-01-19:
### Is your feature request related to a problem?

When using an Idex printer and you want to print a model not designed for multi material, it is really nice to paint them on like Prus slicer has done. However i do not like the slicer at all. This would be awesome to see in Cura.

### Describe the solution you'd like

add multi material paining like Prusa slicer

### Describe alternatives you've considered

Prusa slicer, not even close as good as Cura

### Affected users and/or printers

Anyone using multi material machines or machines with multi material add ons.

### Additional information & file uploads

[image]

Anacardium12, 2023-05-19:
Please add these feature!

psiho051, 2023-05-27:
When does someone think of dealing with this option so that those of us who have idex printers don't have to use a prusa slicer?