Ultimaker/Cura #14335: It wont let me slice my file.

dontWANTtoBEhere, 2023-01-19:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

I try to hit the slice button, it will try to  load then quit saying it failed and there's a bug.
I updated the app.

### Actual results

It keeps doing the same thing.

### Expected results

It works.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No matter what file I use it says the same thing.

Spacenoidx, 2023-01-19:
I have the same problem.

MariMakes, 2023-01-27:
Hey @dontWANTtoBEhere,

Welcome to the Ultimaker Cura Github 🚀 
Sorry to hear you are stuck 😖 

Is it possible that your printjob is located at 0, 0, 0, and it slices if you move it a little away from the center of the buildplate? 
Then it might be a duplicate of https://github.com/Ultimaker/Cura/issues/13643

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-02-11 by github-actions[bot]