Ultimaker/Cura #14337: Cura build from source: CMakePresets.json missing for pyarcus

mjc-506, 2023-01-19:
### Application Version

5.3.0 (git 7ccb361)

### Platform

Linux

### Printer

-

### Reproduction steps

Following the steps to build Cura from source, but after issuing the `conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv` command, the build fails while building pyarcus, reporting that CMakePresets.json is not found.

I've tried clearing the conan cache and redownloading from github, but no success. I don't know conan well enough to troubleshoot further!

### Actual results

Log as follows: (incomplete unfortuantely, not enough scrollback! I'll redirect to a logfile and attach if useful?)
<details>
`pyarcus/5.2.2: Configuring sources in /home/matt/.conan/data/pyarcus/5.2.2/_/_/source/.
pyarcus/5.2.2: SCM: Getting sources from url: 'https://github.com/Ultimaker/pyArcus'
pyarcus/5.2.2: Copying sources to build folder
pyarcus/5.2.2: Building your package in /home/matt/.conan/data/pyarcus/5.2.2/_/_/build/f2ef1d95f57a30d1552d898af2894cc9514968a1
pyarcus/5.2.2: Generator 'VirtualRunEnv' calling 'generate()'
pyarcus/5.2.2: Generator 'CMakeDeps' calling 'generate()'
pyarcus/5.2.2: Generator 'VirtualBuildEnv' calling 'generate()'
pyarcus/5.2.2: Calling generate()
['\n[build-system]\nrequires = ["sip >=6, <7", "setuptools>=40.8.0", "wheel"]\nbuild-backend = "sipbuild.api"', '\n[tool.sip.metadata]\nname = "pyArcus"\nversion = "5.2.2"\nsummary = "Communication library between internal components for Ultimaker software"\nhome-page = "https://github.com/Ultimaker/libArcus"\nauthor = "Ultimaker B.V."\nlicense = "LGPL-3.0"\ndescription-file = "README.md"\nrequires-python = ">=3.10.4"', '\n[tool.sip.project]\ncompile = false\nsip-files-dir = "python"\nbuild-dir = "/home/matt/.conan/data/pyarcus/5.2.2/_/_/build/f2ef1d95f57a30d1552d898af2894cc9514968a1/build/Release"\ntarget-dir = "/home/matt/.conan/data/pyarcus/5.2.2/_/_/package/f2ef1d95f57a30d1552d898af2894cc9514968a1/site-packages"\npy-include-dir = "/home/matt/.conan/data/cpython/3.10.4/_/_/package/9054a82b60899c3ed90865d5b990fd99ef2dc167/include/python3.10"\npy-major-version = 3\npy-minor-version = 10', "\n[tool.sip.bindings.pyArcus]\nexceptions = true\nrelease-gil = true\nlibraries = ['Arcus', 'python3.10', 'protoc', 'protobuf', 'ssl', 'crypto', 'ffi', 'mpdec++', 'mpdec', 'crypt', 'gdbm', 'sqlite3', 'tk8.6', 'tkstub8.6', 'panelw', 'menuw', 'formw', 'ncurses++w', 'ticw', 'ncursesw', 'tinfow', 'lzma', 'tdbcmysql1.1.1', 'sqlite3.30.1.2', 'tdbc1.1.1', 'tdbcstub1.1.1', 'tdbcpostgres1.1.1', 'tdbcodbc1.1.1', 'itcl4.2.0', 'itclstub4.2.0', 'thread2.8.5', 'tcl8.6', 'tclstub8.6', 'fontconfig', 'expat', 'uuid', 'freetype', 'bz2', 'png16', 'brotlidec-static', 'brotlienc-static', 'brotlicommon-static', 'z']\nlibrary-dirs = ['/home/matt/.conan/data/arcus/5.2.2/_/_/package/09d50a5ecc7914f9de0b3fa1696ed1248b8266a2/lib', '/home/matt/.conan/data/cpython/3.10.4/_/_/package/9054a82b60899c3ed90865d5b990fd99ef2dc167/lib', '/home/matt/.conan/data/protobuf/3.21.4/_/_/package/440bc1b350338e220ee6f38abf4bad8023e831ed/lib', '/home/matt/.conan/data/openssl/1.1.1l/_/_/package/2a19826344ff00be1c04403f2f8e7008ed3a7cc6/lib', '/home/matt/.conan/data/libffi/3.2.1/_/_/package/2a19826344ff00be1c04403f2f8e7008ed3a7cc6/lib', '/home/matt/.conan/data/mpdecimal/2.5.0/_/_/package/4f5b86843304ef4d3ee0bf7af3c3de8ee5f4751b/lib', '/home/matt/.conan/data/libxcrypt/4.4.25/_/_/package/2a19826344ff00be1c04403f2f8e7008ed3a7cc6/lib', '/home/matt/.conan/data/gdbm/1.19/_/_/package/6f71f3a61efaf6b8cef674fbf7b24097e056afa0/lib', '/home/matt/.conan/data/sqlite3/3.36.0/_/_/package/668a23a956a1ae835b578205ecc8274cfd29ae40/lib', '/home/matt/.conan/data/tk/8.6.10/_/_/package/b1680222d657c77e2ae6235944e4a03c074eac93/lib', '/home/matt/.conan/data/ncurses/6.2/_/_/package/7e2093608253f4aaa2f5c56670f3fa891c56f0d0/lib', '/home/matt/.conan/data/xz_utils/5.2.5/_/_/package/2a19826344ff00be1c04403f2f8e7008ed3a7cc6/lib', '/home/matt/.conan/data/tcl/8.6.10/_/_/package/95716e737dc568f8c7699d241b52e1051129637d/lib/tdbcmysql1.1.1', '/home/matt/.conan/data/tcl/8.6.10/_/_/package/95716e737dc568f8c7699d241b52e1051129637d/lib/sqlite3.30.1.2', '/home/matt/.conan/data/tcl/8.6.10/_/_/package/95716e737dc568f8c7699d241b52e1051129637d/lib/tdbc1.1.1', '/home/matt/.conan/data/tcl/8.6.10/_/_/package/95716e737dc568f8c7699d241b52e1051129637d/lib/tdbcpostgres1.1.1', '/home/matt/.conan/data/tcl/8.6.10/_/_/package/95716e737dc568f8c7699d241b52e1051129637d/lib/tdbcodbc1.1.1', '/home/matt/.conan/data/tcl/8.6.10/_/_/package/95716e737dc568f8c7699d241b52e1051129637d/lib/itcl4.2.0', '/home/matt/.conan/data/tcl/8.6.10/_/_/package/95716e737dc568f8c7699d241b52e1051129637d/lib/thread2.8.5', '/home/matt/.conan/data/tcl/8.6.10/_/_/package/95716e737dc568f8c7699d241b52e1051129637d/lib', '/home/matt/.conan/data/fontconfig/2.13.93/_/_/package/5eddef872c5e787cdd7e8424a4fbbfff36480650/lib', '/home/matt/.conan/data/expat/2.4.1/_/_/package/a1ba772dbe13186199f7ecf8dde362ba9613ecf7/lib', '/home/matt/.conan/data/libuuid/1.0.3/_/_/package/2a19826344ff00be1c04403f2f8e7008ed3a7cc6/lib', '/home/matt/.conan/data/freetype/2.12.1/_/_/package/ce89339e603b4cf65a2616737fa8a88fe8b27b72/lib', '/home/matt/.conan/data/bzip2/1.0.8/_/_/package/3cfc45772763dad1237052f26c1fe8b2bae3f7d2/lib', '/home/matt/.conan/data/libpng/1.6.37/_/_/package/176676f6669662b0df9dfa3eaa82bc431da39043/lib', '/home/matt/.conan/data/brotli/1.0.9/_/_/package/2a19826344ff00be1c04403f2f8e7008ed3a7cc6/lib', '/home/matt/.conan/data/zlib/1.2.12/_/_/package/2a19826344ff00be1c04403f2f8e7008ed3a7cc6/lib']\ninclude-dirs = ['/home/matt/.conan/data/arcus/5.2.2/_/_/package/09d50a5ecc7914f9de0b3fa1696ed1248b8266a2/include', '/home/matt/.conan/data/cpython/3.10.4/_/_/package/9054a82b60899c3ed90865d5b990fd99ef2dc167/include', '/home/matt/.conan/data/cpython/3.10.4/_/_/package/9054a82b60899c3ed90865d5b990fd99ef2dc167/include/python3.10', '/home/matt/.conan/data/protobuf/3.21.4/_/_/package/440bc1b350338e220ee6f38abf4bad8023e831ed/include', '/home/matt/.conan/data/openssl/1.1.1l/_/_/package/2a19826344ff00be1c04403f2f8e7008ed3a7cc6/include', '/home/matt/.conan/data/libffi/3.2.1/_/_/package/2a19826344ff00be1c04403f2f8e7008ed3a7cc6/include', '/home/matt/.conan/data/mpdecimal/2.5.0/_/_/package/4f5b86843304ef4d3ee0bf7af3c3de8ee5f4751b/include', '/home/matt/.conan/data/libxcrypt/4.4.25/_/_/package/2a19826344ff00be1c04403f2f8e7008ed3a7cc6/include', '/home/matt/.conan/data/gdbm/1.19/_/_/package/6f71f3a61efaf6b8cef674fbf7b24097e056afa0/include', '/home/matt/.conan/data/sqlite3/3.36.0/_/_/package/668a23a956a1ae835b578205ecc8274cfd29ae40/include', '/home/matt/.conan/data/tk/8.6.10/_/_/package/b1680222d657c77e2ae6235944e4a03c074eac93/include', '/home/matt/.conan/data/ncurses/6.2/_/_/package/7e2093608253f4aaa2f5c56670f3fa891c56f0d0/include', '/home/matt/.conan/data/ncurses/6.2/_/_/package/7e2093608253f4aaa2f5c56670f3fa891c56f0d0/include/ncursesw', '/home/matt/.conan/data/xz_utils/5.2.5/_/_/package/2a19826344ff00be1c04403f2f8e7008ed3a7cc6/include', '/home/matt/.conan/data/tcl/8.6.10/_/_/package/95716e737dc568f8c7699d241b52e1051129637d/include', '/home/matt/.conan/data/fontconfig/2.13.93/_/_/package/5eddef872c5e787cdd7e8424a4fbbfff36480650/include', '/usr/include/uuid', '/home/matt/.conan/data/expat/2.4.1/_/_/package/a1ba772dbe13186199f7ecf8dde362ba9613ecf7/include', '/home/matt/.conan/data/libuuid/1.0.3/_/_/package/2a19826344ff00be1c04403f2f8e7008ed3a7cc6/include', '/home/matt/.conan/data/libuuid/1.0.3/_/_/package/2a19826344ff00be1c04403f2f8e7008ed3a7cc6/include/uuid', '/home/matt/.conan/data/freetype/2.12.1/_/_/package/ce89339e603b4cf65a2616737fa8a88fe8b27b72/include', '/home/matt/.conan/data/freetype/2.12.1/_/_/package/ce89339e603b4cf65a2616737fa8a88fe8b27b72/include/freetype2', '/home/matt/.conan/data/bzip2/1.0.8/_/_/package/3cfc45772763dad1237052f26c1fe8b2bae3f7d2/include', '/home/matt/.conan/data/libpng/1.6.37/_/_/package/176676f6669662b0df9dfa3eaa82bc431da39043/include', '/home/matt/.conan/data/brotli/1.0.9/_/_/package/2a19826344ff00be1c04403f2f8e7008ed3a7cc6/include', '/home/matt/.conan/data/brotli/1.0.9/_/_/package/2a19826344ff00be1c04403f2f8e7008ed3a7cc6/include/brotli', '/home/matt/.conan/data/zlib/1.2.12/_/_/package/2a19826344ff00be1c04403f2f8e7008ed3a7cc6/include', 'include']\npep484-pyi = true\nstatic = false\ndebug = false", "\nheaders = ['PythonMessage.h']\nsources = ['src/PythonMessage.cpp']", "\nextra-compile-args = ['-std=c++17', '-m64', '-fPIC', '-O3']\nextra-link-args = ['-m64']"]
[... cut ...]
ERROR: pyarcus/5.2.2: Error in build() method, line 103
        cmake = CMake(self)
        FileNotFoundError: [Errno 2] No such file or directory: '/home/matt/.conan/data/pyarcus/5.2.2/_/_/build/f2ef1d95f57a30d1552d898af2894cc9514968a1/build/Release/generators/CMakePresets.json'
[matt@mattop Cura]$ 
`
</details>

### Expected results

I don't get to activating the virtual environment or running Cura.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

N/A

MariMakes, 2023-01-20:
Hey @mjc-506,

😖 Sorry to hear you are stuck.
Can you help me understand? What do you want to achieve by creating a Build?
Are you just trying to utilize the workaround mentioned here: https://community.ultimaker.com/topic/42417-fix-for-connecting-creality-ender-3-neo-to-mac-via-usb/
Are you intending to share it with others or are you using it for research purposes?
I would love to know more about your project, maybe we can help.

I'm only a fledgling member of the Build From Source Club 👑 
Before we get started, I have a few questions to narrow down the troubleshooting:

1. Can you make sure you have the latest Conan version?  
`pip install conan --upgrade `
You can check the [Conan Documentation]( https://docs.conan.io/en/latest/installation.html ) for the latest version and installation instructions

2. Did you try clearing the Conan Cache? 
`conan remove "*" -f `

3. Can you make sure you have the latest config with this command?
`conan config install https://github.com/Ultimaker/conan-config.git`

4. Can you make sure you have a default profile?
`conan profile new default --detect --force`

5. Can you make sure you have the cura-ce Artifactory repo and not the cura Artifactory repo?
Please check `cura remote list`, this will return your Artifactory Repo
If you see cura in your list, run ` conan remote remove cura`

6. With these steps you should have a clean slate and you can follow the steps on our wiki again. 
[You can find it here. ](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source) 

If you are stuck in any way, please reach out. We are happy to help out. 
The best way to do that is to provide us with your complete run-out log. 
Not just the part that fails, but all the lines generated after you've given your command. 
We need it see your profiles and packages.

mjc-506, 2023-01-20:
Hi,

Thanks for your response. I'm hoping to implement https://github.com/Ultimaker/Cura/issues/5248 but it's not good making changes to the code if I can't compile it in the first case :) Once I can compile the 'stock' code, I intend to make the changes, make sure it compiles/works, and then a couple of PRs to Cura and CuraEngine.

1:
```bash
[matt@mattop Cura]$ pip install conan --upgrade
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: conan in /home/matt/.local/lib/python3.10/site-packages (1.57.0)
Requirement already satisfied: pluginbase>=0.5 in /home/matt/.local/lib/python3.10/site-packages (from conan) (1.0.1)
Requirement already satisfied: bottle<0.13,>=0.12.8 in /home/matt/.local/lib/python3.10/site-packages (from conan) (0.12.23)
Requirement already satisfied: fasteners>=0.14.1 in /home/matt/.local/lib/python3.10/site-packages (from conan) (0.18)
Requirement already satisfied: PyYAML<=6.0,>=3.11 in /usr/lib/python3.10/site-packages (from conan) (6.0)
Requirement already satisfied: urllib3<1.27,>=1.26.6 in /usr/lib/python3.10/site-packages (from conan) (1.26.12)
Requirement already satisfied: python-dateutil<3,>=2.7.0 in /usr/lib/python3.10/site-packages (from conan) (2.8.2)
Requirement already satisfied: tqdm<5,>=4.28.1 in /home/matt/.local/lib/python3.10/site-packages (from conan) (4.64.1)
Requirement already satisfied: pygments<3.0,>=2.0 in /usr/lib/python3.10/site-packages (from conan) (2.14.0)
Requirement already satisfied: requests<3.0.0,>=2.25 in /usr/lib/python3.10/site-packages (from conan) (2.28.1)
Requirement already satisfied: patch-ng<1.18,>=1.17.4 in /home/matt/.local/lib/python3.10/site-packages (from conan) (1.17.4)
Requirement already satisfied: node-semver==0.6.1 in /home/matt/.local/lib/python3.10/site-packages (from conan) (0.6.1)
Requirement already satisfied: PyJWT<3.0.0,>=2.4.0 in /usr/lib/python3.10/site-packages (from conan) (2.6.0)
Requirement already satisfied: six<=1.16.0,>=1.10.0 in /usr/lib/python3.10/site-packages (from conan) (1.16.0)
Requirement already satisfied: colorama<0.5.0,>=0.3.3 in /usr/lib/python3.10/site-packages (from conan) (0.4.6)
Requirement already satisfied: Jinja2<4.0.0,>=3.0 in /usr/lib/python3.10/site-packages (from conan) (3.1.2)
Requirement already satisfied: distro<=1.7.0,>=1.0.2 in /home/matt/.local/lib/python3.10/site-packages (from conan) (1.7.0)
Requirement already satisfied: MarkupSafe>=2.0 in /usr/lib/python3.10/site-packages (from Jinja2<4.0.0,>=3.0->conan) (2.1.1)
Requirement already satisfied: idna<4,>=2.5 in /usr/lib/python3.10/site-packages (from requests<3.0.0,>=2.25->conan) (3.4)
[matt@mattop Cura]$ 
```
So all good there?

2: Yup, conan cache cleared, no change unfortunately (log of attempted build following cache empty attached)

3:
```bash
[matt@mattop Cura]$ conan config install https://github.com/Ultimaker/conan-config.git
Trying to clone repo: https://github.com/Ultimaker/conan-config.git
Repo cloned!
Defining remotes from remotes.txt
Copying file global.conf to /home/matt/.conan/.
Processing conan.conf
Copying file .gitignore to /home/matt/.conan/.
Skip README.md
Copying file VirtualPythonEnv.py to /home/matt/.conan/generators
Copying file GitHubActionsRunEnv.py to /home/matt/.conan/generators
Copying file PyCharmRunEnv.py to /home/matt/.conan/generators
Copying file GitHubActionsBuildEnv.py to /home/matt/.conan/generators
Copying file cura.jinja to /home/matt/.conan/profiles
Copying file cura_build.jinja to /home/matt/.conan/profiles
[matt@mattop Cura]$ 
```
Again, no change.

4: 
```bash
[matt@mattop Cura]$ conan profile new default --detect --force
Found gcc 12.2
Found clang 14.0
gcc>=5, using the major as version

************************* WARNING: GCC OLD ABI COMPATIBILITY ***********************
 
Conan detected a GCC version > 5 but has adjusted the 'compiler.libcxx' setting to
'libstdc++' for backwards compatibility.
Your compiler is likely using the new CXX11 ABI by default (libstdc++11).

If you want Conan to use the new ABI for the default profile, run:

    $ conan profile update settings.compiler.libcxx=libstdc++11 default

[... 78 lines cut ...]
Downloading conanmanifest.txt completed [0.06k]                                          
Downloading conanfile.py completed [0.15k]                                               
uranium/latest@ultimaker/testing: Downloaded recipe revision e384df30d99863acdf2685d88119af96
fdm_materials/latest@ultimaker/testing: Not found in local cache, looking in remotes...
fdm_materials/latest@ultimaker/testing: Trying with 'cura-ce'...
Downloading conanmanifest.txt completed [0.06k]                                          
Downloading conanfile.py completed [0.16k]                                               
fdm_materials/latest@ultimaker/testing: Downloaded recipe revision 4459d79366bda9aaee9e4bef00bf0346
cura_binary_data/latest@ultimaker/testing: Not found in local cache, looking in remotes...
cura_binary_data/latest@ultimaker/testing: Trying with 'cura-ce'...
Downloading conanmanifest.txt completed [0.06k]                                          
Downloading conanfile.py completed [0.16k]                                               
cura_binary_data/latest@ultimaker/testing: Downloaded recipe revision 9dd7d8d0f8d8cd9f75927a79f72cc8fc
pyarcus/5.2.2: Not found in local cache, looking in remotes...
pyarcus/5.2.2: Trying with 'cura-ce'...
pyarcus/5.2.2: Trying with 'conan-center'...
[... cut ...]
ERROR: Unable to find 'pyarcus/5.2.2' in remotes
[matt@mattop Cura]$ 
```
I also note that `conan config install https://github.com/Ultimaker/conan-config.git` adds `cura` back in to conan's remotes. There's no mention on the wiki of removing `cura` so presumably this isn't required any more?

Here's a log of the build, following a cleared cache, fresh download of the conan profile (but not removing `cura` from conan remotes)

[buildlog.txt](https://github.com/Ultimaker/Cura/files/10466899/buildlog.txt)

gconcha1, 2023-02-01:
Hi! :) 

I'm having the same issue, did you manage to solve it?

MariMakes, 2023-02-01:
Hey,

I had one of the Devs take a look. Can you try the following things? 

Can you make sure you have specifically Conan version 1.56.0?  
`pip install conan == 1.56.0 --force `
Apparently, there is a bug in higher versions that I wasn't aware of. 

Please check if the cura remote is specified with the following command:
` conan remote list `

If you only see the `cura-ce` remote please re-add the `cura remote`:
` conan remote add cura https://ultimaker.jfrog.io/artifactory/api/conan/cura-internal `

gconcha1, 2023-02-01:
Hi again!

Thanks for the quick response. 

I tried that and another error poped up, I don't know what I'm doing wrong. 

Sorry if I'm being dumb, it's my first time trying to run a program from source.

[image]

mjc-506, 2023-02-01:
Thanks @MariMakes 
`pip install conan == 1.56.0 --force` worked for me!

Happy to close if @gconcha1 's error is not related. But worth updating the build instructions?

gconcha1, 2023-02-02:
Hi! 

You're free to close, I finally did it. It worked.

Thanks @MariMakes  and @mjc-506 .

MariMakes, 2023-02-03:
Ah, so glad to hear ❤️  Good luck 💪 

I've updated the documentation and we have a ticket to expand support to the newer Conan versions.

closed 2023-02-03 by MariMakes

real-artswan, 2023-02-04:
To me a correct command for pip is
`pip install --force-reinstall -v "conan==1.56.0"`

ZZQiang-An, 2023-02-21:
[image]
Can someone please come and stick me or tell me what to do?