Ultimaker/Cura #14343: Bad slicing glitches/visuals on linux with appimage

DuckersMcQuack, 2023-01-20:
### Application Version

4.12.1

### Platform

Steam deck/steamOS/arch with KDE plasma

### Printer

custom

### Reproduction steps

Run arch with KDE plasma, download 4.12.1 appimage, and start a slice.

### Actual results

Visuals on slices are broken, they print fine, i just can't effectively inspect slice on the steam deck.

[image]

### Expected results

Visuals to have clean render of the slice.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10468572/cura.log)

snoopen, 2023-01-25:
Can confirm this also affects me.

First slice it appears fine, but any interaction (to rotate or move model/view, layer slider, etc) all breaks the render. However top most layer (as changed with the slider) always appears fine, just all the other layers render incorrectly.

Tested with all Appimages 5.3.0-alpha back to 4.131 (modern where available)

ThinkPad T14 Gen 3
Pop!_OS 22.04 LTS x86_64
Gnome 42
OpenGL version string: 4.6 (Compatibility Profile) Mesa 22.2.0

MariMakes, 2023-01-27:
Hey All,

Thanks for your report 👍
If the preview is not showing up as expected is often related to a driver issue. 

This issue reminds me of https://github.com/Ultimaker/Cura/issues/14324
Could it be a duplicate?

DuckersMcQuack, 2023-01-27:
As in same user? Nope. Mine was a ender 3 pro, but not considered anymore as i replaced board, firmware with klipper, lead screw for belts, stepper motors, hotend and extruder. And i won't use cura 5 because i don't like it's scrolling speeds, plus the addons i use isn't supported/ported to 5 yet. But the other user uses 5, so i guess it's an issue that has followed since 4 for linux/lack of drivers for amd for linux to render slice properly perhaps.

But it might be a driver related issue. Cause i don't know if steam deck has any drivers? Maybe barebone ones to communicate with linux. Cause these issues are not present on windows. As i use 4.12.1 on windows as well, but nvidia gpu.

MariMakes, 2023-01-30:
Hey @DuckersMcQuack,

I apologize, I completely misread that you are running Cura on a Steam Deck!
I haven't seen reports on that device, yet! Unfortunately, is currently not a supported Operating System. 
That explains the reason why you might be missing certain device drivers. 

You can learn more about which operating systems we support here. 
https://support.makerbot.com/s/article/1667337917781

Perhaps you can crowdsource your answer [on our community forum here](https://community.ultimaker.com/)?

Since this is an issue tracker, I'll close this issue since it's not something we plan to solve on our side soon. 
I hope you understand.

closed 2023-01-30 by MariMakes

snoopen, 2023-01-31:
@MariMakes fyi SteamOS is a linux distribution.

DuckersMcQuack, 2023-02-04:
> Hey @DuckersMcQuack,
>
> I apologize, I completely misread that you are running Cura on a Steam Deck! I haven't seen reports on that device, yet! Unfortunately, is currently not a supported Operating [...]

Hehe no worries. I can print just fine in the meantime, just can't inspect slice :P Just until i get my main rig running again. Hopefully the steam deck will get drivers to properly show the slice.