Ultimaker/Cura #14344: Add `Print Speed` (and maybe other speed related settings like `Support Speed`) to Material Print Settings

CoryCharlton, 2023-01-20:
### Is your feature request related to a problem?

I'm always frustrated when I have to create a duplicate profile for a specific material. This results in me having to select both the correct material and correct profile. Additionally I need to update multiple profiles if I want to change a setting that is not specific to a given material.

### Describe the solution you'd like

I would like more settings to be added to the Material Print Settings section. In particular the "speed" related settings can be specific to a given material (ex: I like to print slower when using PETG/TPU than I would with PLA) where as other profile settings (such as `Wall Line Count`) may be the same across any material.

### Describe alternatives you've considered

As mentioned in the description of the problem I am currently duplicating profiles for specific materials. For instance I have a `Good (0.20)` profile and a  `Good (0.20) PETG` profile. All the settings across these profiles are identical other than `Print Speed` and `Support Speed`. 

This is not working out well because I have to remember to select both the PETG material and the PETG specific profile. Additionally if I decide that my `Good (0.20)` should have a different value, say increasing `Wall Line Count` from 3 to 4) I need to make sure to update all the material specific profiles as well.

Another alternative would be if there could be a master profile that other profiles inherited default values from. In my example the `Good (0.20) PETG` profile would inherit from the `Good (0.20)` profile and only the `Print Speed` and `Support Speed` values would be stored in the `Good (0.20) PETG` profile. I feel this isn't the best solution because it still means I need to select both a material and a material specific profile which negates the purpose of having a material at all.

### Affected users and/or printers

Everyone will benefit from these changes.

Asterchades, 2023-01-21:
There's a plug-in called "Material Settings" which lets you choose whatever settings you please to add to the material properties. This even includes settings added by other plug-ins, such as Z-offset, ArcWelder, or Linear Advance.

CoryCharlton, 2023-01-21:
> There's a plug-in called "Material Settings" which lets you choose whatever settings you please to add to the material properties. This even includes settings added by other plug-ins, such as Z-offset, ArcWelder, or Linear Advance.

Thanks so much for pointing me to that plug-in I'll start using it now!

That being said I think this should be core functionality in Cura so I'll leave this request open unless you guys feel it should be closed.