Ultimaker/Cura #14345: FILL-IN ERRORS IN GCODES

DalSheron, 2023-01-20:
### Application Version

5.1.0

### Platform

Windows 10 Pro

### Printer

TwoTrees SP-5

### Reproduction steps

I write these mugs from DEEP ROCK GALACTIC so I can print them, Rotate them by 90 degrees so bottoms meet print bed, then scale them to 500%, write to GCODE and then the tops are filled in where they shouldnt be.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Actual results

Mug internal surfaces are joined by layers of material that should not be there, rendering mugs useless from an appearance perspective

### Expected results

The mug internals should be voided as they are in the preview models.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

these are the files I used
[DRG-Beer-mugs.zip](https://github.com/Ultimaker/Cura/files/10469223/DRG-Beer-mugs.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/10469229/cura.log)

GregValiant, 2023-01-20:
Thanks for the report.
I thought for sure I'd be telling you to turn off "Remove All Holes".  I still might but here is a view of the first model I opened...
[image]

All of those red areas are problems within the model.  I tried to repair that one in MS 3D Builder but it didn't work.  Mesh Mixer couldn't fix them either  This is the "deep" side.
[image]

I opened a couple of others and they also show errors.  When Cura paints something in polka-dots it indicates problems in the model.  I wouldn't expect these models to slice correctly.
I'll remove the bug label from this but I'll leave it open so you can respond.  Those are some bad models.  Even the ones that looked like they sliced OK have gaps and holes.

Asterchades, 2023-01-21:
Videogame models are very rarely, if ever going to be 3D printable without modification. They're only really made to look good on a screen - due to things like Z-buffering and backface culling there are a lot of modelling shortcuts which can, and often are, used to both reduce the artist's work and to reduce the GPU load (clipping is less expensive than additional geometry). And that's before considering geometry which is only used to apply effects (glows, liquids, etc).

These in particular appear to be riddled with internal geometry - to the point where it's going to require largely manual correction, given Greg's experiences with automated tools.

DalSheron, 2023-01-22:
too bad I have zero experience with modeling outside of Autodesk Inventor, otherwise 3d modeling is largely alien to me.

GregValiant, 2023-01-22:
@DalSheron  if you really want those models you could use a utility like [THIS ONE](https://www.cadforum.cz/catalog_en/stl2dwg.asp) and use their STLtoDWG tool.  It's time consuming to do it that way as you are basically starting from scratch.  You will end up with a good model though.
I'm going to go ahead and close this as it isn't a Cura problem.  I hope you understand.

closed 2023-01-22 by GregValiant