Ultimaker/Cura #14351: Slice fails after loading large or more complex stl

OzButcher78, 2023-01-21:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Anycubic Vyper

### Reproduction steps

Imported a .STL
Resized  206.36%
Sliced which failed. 

### Actual results

Slice fails and displays error message to report the bug

### Expected results

Slice would complete 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

This is the .STL I tried to slice at 206%

[Garden_Turtle_421809.zip](https://github.com/Ultimaker/Cura/files/10472386/Garden_Turtle_421809.zip)
[AV_turtle-project-3mf.zip](https://github.com/Ultimaker/Cura/files/10472392/AV_turtle-project-3mf.zip)
[cura.log.zip](https://github.com/Ultimaker/Cura/files/10472398/cura.log.zip)

GregValiant, 2023-01-21:
Thanks for the report and the files.
This sliced for me on the first try in both 5.3 and 5.2.1.  I'm on Windows 11.
There have been many reports of a bug that can keep good models from slicing when they are in the middle of the Cura build plate.  You can try moving and/or rotating the model.  Sometimes that fools the bug.
I'll add the slicer error to this and someone from the Cura team will take a look.

OzButcher78, 2023-01-22:
This happened with several .stls i loaded. but I noticed I had a ram error on my pc. once I rebooted everything worked again.

GregValiant, 2023-01-23:
Oh, hardware issue.  I'll go ahead and remove the bug label and close this then.  Thanks for getting back on this.

closed 2023-01-23 by GregValiant