Ultimaker/Cura #14352: Slicing fails

Filamentuser, 2023-01-21:
### Application Version

5.2.1

### Platform

windows11

### Printer

flsun sr

### Reproduction steps

tried to clice

### Actual results

[image]

### Expected results

produced gcode

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

MariMakes, 2023-01-27:
Hey @Filamentuser,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Is it possible that your printjob is located at 0, 0, 0, and it slices if you move it a little away from the center of the buildplate? 
Then it might be a duplicate of https://github.com/Ultimaker/Cura/issues/13643

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

Filamentuser, 2023-01-27:
moved location on build plate and successfully slices.

MariMakes, 2023-01-27:
Thanks for getting back to us. 

Then it looks like a duplicate of https://github.com/Ultimaker/Cura/issues/13643
Luckily we seem to have found a solution direction for that issue this morning, so it is probably going to be fixed int he next release. 

Glad to hear you are unstuck. 
I'll close the issue.

closed 2023-01-27 by MariMakes