Ultimaker/Cura #14355: Text is cut off in on-screen help

bellzw, 2023-01-21:
### Application Version

5.1.0

### Platform

64 bit Windows 10 v. 21H2 build 19044.2486 running on a Lenovo Yoga 14

### Printer

modified BIBO (I added overrides for actual max speeds, accelerations, etc.)

### Reproduction steps

Hover over setting (Z seam Alignment is one)

### Actual results

If a sufficiently large help window opens, the bottom line or lines are cut off.  What's actually cut off seems to depend on the monitor resolution.  File names indicate monitor resolutions.  The 2560x1440 file is model S3222DGW.  The 1980x1080 file is a Dell S2309Wb.

[image]
[image]

### Expected results

Expected to be able to scroll down to see all the text.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I have Settings Guide 2.9.2 installed.

fieldOfView, 2023-01-23:
It should be possible to scroll the tooltips with the mousewheel or two-finger gesture on a touchpad. Could you let us know if that works for you?

Sonicboom247, 2023-01-23:
i love seeing the settings, but sometimes i don't need all of them, i would love to see a "Don't show again" for some of the tooltips, I would also love to be able to resize or move the window for some of these, this is especially true when working on machine settings, the settings pop up on top of the window and block anything you are doing, so if you are trying to tweak as you go, it is impossible.

bellzw, 2023-01-24:
I'm using Cura 5.1.0 on a laptop whose native display is touch sensitive, and have two other monitors which are not.  On the laptop's display, dragging with my fingers works.  However, on the two other displays (the screenshots above), and on the laptop's display, the wheel is intermittent at best.  If I spin it fast, sometimes the display moves up and shows me the last line.  If I spin it slowly, like 1 or 2 clicks at a time, the display remains with the last line cut off.  

When moving the text with the mouse wheel, a grey scroll bar appears at the right of the window.  The mouse cannot capture the grey scroll bar regardless of which display Cura is on.