Ultimaker/Cura #14359: slicing error

K3DPrints, 2023-01-22:
### Application Version

5.0

### Platform

windows 10

### Printer

Kobra Plus

### Reproduction steps

1.uploaded file from google drive, 
2.sliced with my normal settings
 

### Actual results

slicing failed.
slicing failed with an unexpected error

### Expected results

my file should be sliced?

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Valentine_Pikachu02.zip](https://github.com/Ultimaker/Cura/files/10475101/Valentine_Pikachu02.zip)

Spacenoidx, 2023-01-23:
> Thanks for the report and the file. This sliced for me on the first try so I am unable to duplicate the problem. There is a bug that can keep models from slicing when they are centered on the Cura bed. Try moving the model and/or rotating it and see if it makes a difference.
> 
> [image]

I consistently try this and it does not work.

tomblenz, 2023-02-15:
FWIW, I had a similar issue and nudging it a few mm's back and rotating slightly fixed whatever was going wrong and it was able to slice.