Ultimaker/Cura #14360: Bug experienced while slicing

geraldchus, 2023-01-22:
### Application Version

5.3.0 Alpha-Xmas

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

Load file from disc.
Slicing worked to about 50% completion then timed out.

### Actual results

Slicing runs forever

### Expected results

finish slicing to ready print
[CE3E3V2_Owl_Facing_Left_fixed_sc.zip](https://github.com/Ultimaker/Cura/files/10475137/CE3E3V2_Owl_Facing_Left_fixed_sc.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Owl_Facing_Left_fixed_sc.zip](https://github.com/Ultimaker/Cura/files/10475129/Owl_Facing_Left_fixed_sc.zip)

DerGenaue, 2023-02-02:
Please check whether disabling limit branch reach solves your problem;
Then this would be a duplicate of this:
https://github.com/Ultimaker/Cura/issues/14133

MariMakes, 2023-04-05:
Hey All,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Thanks for the help @DerGenaue!

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/14133
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/14133