Ultimaker/Cura #14362: Support blockers do not work (5.1.0)

skiddyfisk, 2023-01-23:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ender 3 pro

### Reproduction steps

attempted to add a support blocker to a project

### Actual results

literally nothing, the supports continue to be generated within the area that should have been blocked, resulting in auto-generated supports that will be impossible to cleanly remove from the printed model

### Expected results

there should be no supports within the space occupied by the support blocker cube

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

this has been an issue for several versions now, fix it already, this is ridiculous
[support blockers dont work.zip](https://github.com/Ultimaker/Cura/files/10476492/support.blockers.dont.work.zip)

GregValiant, 2023-01-23:
Thanks for the report.
My printer is also an Ender 3 Pro.  I've never had an issue with the Cura support blockers.  I have noticed that below 5mm in the X and/or Y they don't work.  Here I've added a blocker on the head.
[image]

The model has issues.  The "rebwok z6.stl" model is not watertight.  That may be having some effect on the support blocker as well.

Here you can see right through the firearm.  I don't know if that will print.  It might be because I have a blocker in place and so there is no support on the interior of the barrel.
[image]

Here is one of my own projects with 9 support blockers in it.  This printed correctly and all the blockers worked as I expected.
[image]

This is one of those things where if it was a problem then people wouldn't use Cura.  I'm not quite sure what to make of this.
I'll leave the bug label on it and someone from the Cura team will take a look.

skiddyfisk, 2023-01-23:
I'm glad they work for you. Not sure what bearing that has on them not working for me. Thanks for your input

GregValiant, 2023-01-23:
There are certain combinations of settings that don't work well together.  Try making the support blocker a minimum of 5mm in the X and/or Y.  There doesn't seem to be a minimum in the Z.
When I resized your blocker it worked.  Here you can see there is no support for the hat brim.
[image]

skiddyfisk, 2023-01-24:
Thanks for confirming the bug. Interesting that it only seems to happen under 5mm. Hopefully they get this fixed promptly.

Asterchades, 2023-01-24:
Make sure that your Support Join Distance actually allows for gaps that small between support structures. This should default to 2mm but if it's been increased it could easily result in the structures either side of the blocker being linked together _through_ the blocker until it reaches 5mm thick.

tlhintoq, 2023-02-26:
Pretty sure that's not it.  Because these blockers are WAY over some tiny default join distance and still they aren't blocking anything.
Just a thought...While building the tree if you hit a blocker: Stop.
Treat it like a force field.  Build straight up against it until you go over it or hit the model, whichever comes first.

[image]

tlhintoq, 2023-03-03:
[image]

Asterchades, 2023-03-03:
This latest image looks like the supports are towers which, despite not actually getting there, are _trying_ to support the top of those small indents just above the blockers. Extending the blockers up or disabling Support Towers may remedy this particular situation.

Note that there is work currently being done (or has been done) on towers leading to nothing. The available 5.3.0 beta (not to be confused with the alpha featuring new tree supports) indicates as much, in addition to other tweaks and fixes to support generation. It could be worth checking this most recent model with that build to see if it's improved.

I'm unsure as to what's happening with the Bullseye duct. But then I've always felt that trees were moved from experimental prematurely, so it doesn't come as a great surprise to me that they're misbehaving. It's _possible_ they're still being generated in an effort to reach the top of the internal duct cavity (despite not actually being able to reach it), which would mean the blockers need to extend upwards further.