Ultimaker/Cura #14368: Layers start mid-air

H0RU52, 2023-01-24:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Artillery Sidewinder X1

### Reproduction steps

1. I sliced the file.
2. Sad panda.

### Actual results

Layers start in mid-air

### Expected results

Layers do not start in the air.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[layers-mid-air.zip](https://github.com/Ultimaker/Cura/files/10486756/layers-mid-air.zip)
[image]

smartavionics, 2023-01-24:
Does setting the overhanging wall angle to a value < 90 help?

MariMakes, 2023-02-03:
Hey @H0RU52,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Wow, that's a super interesting model 🤩 
This issue reminds me of https://github.com/Ultimaker/Cura/issues/9369

The suggestion from @smartavionics seems like a good one. Did you get a chance to try it? 
Did you try seam corner preference, shortest? 

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-02-17 by github-actions[bot]