Ultimaker/Cura #14369: 5.3.0 Sliced prints force restart of Marlin 2.1.2/SKR Mini V1.3 Mid print while using RepetierServer

Xgl0baltk, 2023-01-24:
### Application Version

5.3.0 Alpha

### Platform

Windows 10

### Printer

Ender3 V2

### Reproduction steps

Sliced different files using my standard profile which was taken over from 5.2.1 and 5.1.1 where it worked great.

no out of the ordinary settings, all apps are uptodate

Gcode was sent to printer through RepetierServer - USB connection - on a RaspberryPi 4

### Actual results

The print stops midprint, if i print the same file, it stops at the same exact point, restarts the printer which gets stuck on the Marlin 2.1.2 Boot screen and does not go on from there.

Surface finish and overall print quality work great, it simply stops and restarts the printer.

### Expected results

Slicing the exact same file with the exact same profile and settings in 5.2.1 - it prints and finishes the print and parks when done as it is supposed to, no hiccups, good results.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3E3V2_Griff_Chuck2.zip](https://github.com/Ultimaker/Cura/files/10488245/CE3E3V2_Griff_Chuck2.zip)

added both project files from 5.3.0 and 5.2.1 in case it makes a difference, as well as the Gcodes from both versions, sliced with the projet files accordingly

GregValiant, 2023-01-24:
Thanks for the report and the files.
Comparing two gcode files isn't easy.  What I decided (for the first pass) was that:
G0 lines and G1 lines probably weren't a problem.
M204 S1100 and M204 S800 probably weren't a problem.
Comments aren't a problem.
So what's left...

**5.3 gcode**.................................................**5.2.1 gcode**
M140 S75..................................................M140 S75
M105...........................................................M105
M190 S75..................................................M190 S75
M104 S210................................................M104 S210
M105...........................................................M105
M109 S210................................................M109 S210
M82 ;absolute extrusion mode.........M82 ;absolute extrusion mode
G92 E0 ; Reset Extruder........................G92 E0 ; Reset Extruder
G28 ; Home all axes...............................G28 ; Home all axes
G29...............................................................G29
G92 E0 ; Reset Extruder........................G92 E0 ; Reset Extruder
M83 ;relative extrusion mode...........M83 ;relative extrusion mode
M107...........................................................M107
**End of the startup gcode**
M205 X12 Y12.........................................M205 X12 Y12
M106 S85..................................................M106 S85
M106 S170................................................M106 S170
M106 S255................................................M106 S255
M107	
M106 S255	
M107	
M106 S255
**Start of the Ending Gcode**	
M140 S0.....................................................M140 S0
M204 S500................................................M204 S500
M205 X10 Y10..........................................M205 X10 Y10
M82 ;absolute extrusion mode.........M82 ;absolute extrusion mode
M107............................................................M107
G91 ;Relative positioning.....................G91 ;Relative positioning
G90 ;Absolute positioning...................G90 ;Absolute positioning
	
M106 S0 ;Turn-off fan...........................M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend...................M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed..........................M140 S0 ;Turn-off bed

So throwing out the extrusion and travel lines, all the lines that are comments, and the "known" accel lines, that's all that is left.  The 5.3 file has the fan turning on and off twice in the middle of the file and that isn't a normal Cura thing.  Did you post-process or hand edit that file?
The maximum moves of the G1 and G0 lines are within the boundaries of the printer so the print head doesn't appear to have smacked into the endstops.
Support and Support-Interface were turned on in the 5.3 file but not in the 5.2.1 file.  We may be talking apples and oranges here.

Other than the Support - the only difference I see in the files occur above Layer:640 where the bridging appears to be different.  From Layer:0 to layer:640 there is no functional difference in the files.  There are no "odd" comments, and no "odd" moves.

I'm goin to remove the bug label on this.  If you think that is premature let us know and it can be put back on.
If you can think of another analysis I could perform then let me know.  Right now I don't see anything in either file that would stop a print.

You have a third party printer, with a fourth party mainboard, customized by a fifth party and running a sixth party print server through a seventh party RaspberryPI board.  It's possible the issue may lay in there somewhere.  USB printing can be dicey.  Have you tried printing via the SD card?

Xgl0baltk, 2023-01-24:
I know the third, fourth, fifth party setup is an issue, that is why i was hesitant to post it.
But as it happened multiple times in a row, it could have been relevant to report.
Thank you very much for the analysis.
The Gcodes were not processed in any way, they were printed as is. 
Layer 640 actually makes sense, will measure the failed print

have not yet tried it through SD card, will try and see if it stops there and update when done

Sophist-UK, 2024-03-25:
In my experience, this is much more likely to be a Marlin issue than invalid Gcode being produced by Cura.

In particular, as I can testify from personal experience, Marlin can reboot if it runs out of SRAM for the stack. Most memory usage in Marlin is pre-allocated at firmware build time, and the SRAM that is left over after this is then used by the stack, but if you have too many options enabled in Marlin and an older AVR type controller with limited SRAM (which can be as low as 8K!!!) then you can be short on stack memory and this can cause random reboots when the memory gets exhausted (or other symptoms - hangs, thermal runaway (if it hangs with the heater on), multiple beeps). 

The fix is to reduce the pre-allocation of SRAM by turning off or limiting those options that use a lot. The ones I turned off were SD filename sorting, and changing the parameters that affect Input Shaping memory usage (reducing the maximum XY feedrates, increasing the minimum XY Input Shaping frequency).

If the Marlin build you are using was created by you, this should be reasonably easy. If it is a standard Ender build, then try seeing if Ender support can help. Or perhaps you can find help on the Marlin Discord server.)

(If you get very stuck, contact me on the Marlin Discord server and I will see what I can do to help you.)