Ultimaker/Cura #14371: Italian translation in French language

Thornin, 2023-01-24:
### Application Version

5.2.1

### Platform

Windows 10 Windows 11

### Printer

all

### Reproduction steps

1. put language in French
2. read menu on interface, some texte are in Italian

### Actual results

Some text are in Italian instead of French

### Expected results

French text

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

MariMakes, 2023-02-03:
Hey @Thornin,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Oh No! That looks like an error indeed. 
[I checked the translation ](https://github.com/Ultimaker/Cura/blob/8134cce3b8e753a693ff88b09425fd05a1eb7530/resources/i18n/fr_FR/cura.po#L5971) for the next release of Cura and that one looks fine. 

We had to release a 5.2.1 because in 5.2.0 [the translations were incorrect](https://github.com/Ultimaker/Cura/releases/tag/5.2.1) Italian translations were French. In our efforts to resolve it this one might have slipped through. [We put extra effort into](https://github.com/Ultimaker/Cura/pull/14340)  improving the translation process to make sure it's less likely that a human error is introduced. 

Apologies for any inconvenience. 
I'll close the issue since it looks like the translations will be correct in the next release. 
I hope you understand

closed 2023-02-03 by MariMakes

MariMakes, 2023-02-06:
Oh, I apologize I want to give the proper credit. The issues you mentioned have been resolved by one of our community members in this pull request: https://github.com/Ultimaker/Cura/pull/13847

Thank you @xommit 🎉