Ultimaker/Cura #14372: Kenneth

Blockbusted, 2023-01-24:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ultimaker3

### Reproduction steps

Changed the hight from 10. to 6mm

### Actual results

Selected the slicing command and then said report a 

### Expected results

Selected the slicing command Then it said report a 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

File 34 mb zipped

Asterchades, 2023-01-25:
You haven't actually attached either file you've indicated - log or project. Between that and the vague, truncated nature of your report there's basically nothing that can be done in terms of investigation.

Please try again to attach the files. Note that they cannot be attached using the e-mail response system.

Blockbusted, 2023-01-25:
Hi thanks for the reply.

 

It wont let me attach either way email , or log formats even zipped .

I cant seem to see the messages in the red dot on  the market place neither ….

Its not a may ger issue .

 

Thanking you kindly

Kenneth .

MariMakes, 2023-02-03:
Hey @Blockbusted,

@Asterchades is correct, we are missing the correct file to troubleshoot. 

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-02-17 by github-actions[bot]