Ultimaker/Cura #14378: Slicing failed with an unexpected error.

astead, 2023-01-25:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Creality Ender-3 Pro

### Reproduction steps

1. Load STL
2. Hit Slice
Also tried moving model off the center.

### Actual results

Get error message.

### Expected results

Finish slicing.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CE3PRO_B7.zip](https://github.com/Ultimaker/Cura/files/10495469/CE3PRO_B7.zip)
[cura_log.zip](https://github.com/Ultimaker/Cura/files/10495501/cura_log.zip)

I've been trying to build Cura from source to be able to help debug, but get stuck when it says it doesn't recognize sip-build, even though I've done pip install sip...

astead, 2023-01-25:
I've also checked the model with the mesh tools and confirmed the model is watertight.

astead, 2023-01-25:
[image]

MariMakes, 2023-02-03:
Hey @astead,

Thanks for your report 👍
I'm getting a similar error in Cura 5.2.2.

I did notice that the file can be sliced in Cura 4.13.1, perhaps it's related to the variable line width Engine in Cura 5.x
I hope at least it helps you get unstuck. 
[image]

I'll have someone from the team take a look.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker