Ultimaker/Cura #14381: Version 5.2.1 - Printer Nozzle Freezes at 0,0,0

mdjontz, 2023-01-25:
### Application Version

5.2.1

### Platform

Mac

### Printer

Ender 3 V2

### Reproduction steps

1. Download new Cura Version (5.2.1)
2. Sliced STL file with same settings used on previous Cura version.
3. Download GCODE
4. Print GCODE

### Actual results

1. At the beginning of prints (directly prior to the nozzle purge) the printer nozzle freezes at home (0,0,0).

### Expected results

1. Printer would successfully print the output GCODE file.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

While troubleshooting, I tried printing GCODE files generated from the previous version of Cura before updating and they printed successfully.

GregValiant, 2023-01-26:
Thanks for the report.
Without a project file I'd just be guessing.  My guess would be that Creality or a community member changed the definition file you are using.  Maybe there is a command in the StartUp gcode that your printer doesn't like and it's causing the firmware to hang, or wait forever because the command wasn't understood.
You can check your StartUp gcode in the "Manage Printers / Machine Settings".  You can also compare the beginning of one of the gcodes that prints, to one of the ones that doesn't.
Ultimaker maintains Cura but has nothing to do with other manufacturers definition files.  They are submitted by manufacturers or members of the community.

I'll leave the bug label on this for now but it sounds like a printer problem.  My printer is an Ender 3 Pro and I've had no issues with prints stopping.  If you would zip and post a project file then an investigation might show what's going on.

mdjontz, 2023-01-31:
I've attached a Zip File below containing two project files of the same sliced STL from Cura. One is from the previous Cura version that works. The file with "_NEW" in the name is from the 5.2.1 version and is having the issue of freezing at 0,0,0. 

Is there any way I can just copy the start G-code from a working file into my Cura Start G-code settings?

[ProjectFiles.zip](https://github.com/Ultimaker/Cura/files/10548996/ProjectFiles.zip)

GregValiant, 2023-01-31:
The only real difference is that the new gcode has a line "M104 S175" that sets the hot end to the Standby Temperature.  That was added to the definition file because so many printers are coming with ABL installed.
You can go to "Manage Printers" / "Machine Settings" and in the StartUp Gcode box you will see either...
M104 S{material_standby_temperature}
or
M104 S175
You can put a semi-colon in front of it to turn it into a comment.  I really don't see anything in the gcode that should cause the printer to freeze up.

I have a 3 Pro and this is what I use as the Starting Gcode.  It's real close to what you have in the "new" file other than that pre-heat line.
;
; Ender StartUp
;
G90
G21
M82
M140 S{material_bed_temperature_layer_0} ;start the bed
M109 S{material_print_temperature_layer_0} ;heat the hot end
M190 S{material_bed_temperature_layer_0} ;wait for bed
G92 E0 ; Reset E
G28 ; Auto-Home
M220 S100 ;reset feedrate
M221 S100 ;reset flowrate
G1 F2700 Z10.0 ; Move Z
G0 F5000 X1.0 Y20 Z0.3 ; Move
G1 F1500 X1.0 Y200.0 E15 ; First purge line
G0 F5000 X3.0 Y200.0 ; Move
G1 F1500 X3.0 Y20 E30 ; Second purge line
G1 F1800 E28 ;2mm retraction
G92 E0 ; Reset Extruder
G1 F2700 Z12.0 ; Move Z Axis up
G4 S1 ;wait for string to get hard
G0 F5000 X3.0 Y35 Z.35 ; Move over
G0 F5000 X3 Y50 Z0.35 ;wipe
;
;End of StartUp
;